ccrisan/motionpie #293: Can't add IP cameras

brunomariani, 2015-09-07:
[image]

Hi!

I've tried to add 4 IP cameras from different brands (Tenvis, EasyN, Suneyes, Wanscam), but none of them was accepted. It says: "not a supported network camera".

What kind or models of IP cameras are supported?

Regards,

Bruno.

ccrisan, 2015-09-08:
All IP cameras that speak MJPEG or JPEG are supported. Soon there will be support for RTSP cameras. Your problem is most likely the fact that the URL you entered is not the (M)JPG URL of your camera but rather its web interface URL. You need to look the MJPEG URL up online to find it.

closed 2015-09-08 by ccrisan

xxllo, 2016-02-23:
Have an IP Cam from ESCAM qd300 and can not do to activate the cam in Monionpie it . What am I doing wrong???