ccrisan/motionpie #294: Reset date

andrew81, 2015-09-08:
Date in motion pie has reset. Why?
Thanks

ccrisan, 2015-09-08:
More details? Reset to what? 1970? Check the internet connectivity on your motionPie.

andrew81, 2015-09-08:
Yes 1970. I see via internet... And sometimes camera disconnected about 5 second... May be it's problem camera

ccrisan, 2015-09-08:
When your PI reboots (for whatever reason) it starts with 1970 as system date. It then uses the internet connection to retrieve the current date (HTTP on port 80). If that fails, your system time stays at 1970.

closed 2015-09-08 by ccrisan

drknstrmyknight, 2015-09-08:
There is no real-time clock or battery backed CMOS on a Raspberry, so when it is powered on the operating system has no way of determining the time or date. Motionpi will set the time and date from the internet if a connection is available.There are RTC hardware modules available if you have to run off-line. 

   
Reopened #294.—