ccrisan/motionpie #295: bcm2835-vl2: error 0 waiting for frame completion

rockman55, 2015-09-10:
Hello Calin:
 I am running MotionPie version 20157019 on Raspberry Pi 2 and keep getting this error ( when I have a monitor connected). I did not notice this error until I connected a monitor. I did notice that the frames for the individual cameras were only visible momentarily then would go away and were replaced with the icon of a camera with a line running through it. I have changed cameras thinking that that was the problem, but no luck still receiving same error. Can you give me any suggestions?
Thanks,
Danny

ccrisan, 2015-09-10:
You can ignore the error. It's printed out whenever motion is restarted. The reason why it's restarted we have yet to find. Please send me the log files.

rockman55, 2015-09-10:
Calin here are the log files for my system.

Thanks,

Danny

 

 

Boot log: http://s000.tinyupload.com/index.php?file_id=26758650099979798524

Dmesg Log:  <http://s000.tinyupload.com/index.php?file_id=68756929427856958683> http://s000.tinyupload.com/index.php?file_id=68756929427856958683

Messages Log:  <http://s000.tinyupload.com/index.php?file_id=05455604873465025456> http://s000.tinyupload.com/index.php?file_id=05455604873465025456

Motion Log:  <http://s000.tinyupload.com/index.php?file_id=60836381369909922389> http://s000.tinyupload.com/index.php?file_id=60836381369909922389

Motion Eye Log: http://s000.tinyupload.com/index.php?file_id=36447785595264594793

ccrisan, 2015-09-12:
Hmm... hard to tell what's going on, even after looking through the logs. May it be a power problem? An idea would be to remove any unnecessary devices connected to USB and make sure the power adapter is good enough. Have you overclocked your PI?

It would help me if you enabled the debug mode in Expert Settings, reboot the PI and then collect a new set of logs, of course *after* the problem appears.

rockman55, 2015-09-17:
Calin here are the new log files with debug enabled. I have reinstalled the image and set up all my cameras again, and have not had the problem occur. I have a 2.1 amp power supply and the only additional USB device connected is a thumbdrive.

Log files:

Boot: http://s000.tinyupload.com/index.php?file_id=70648589968200390670

Dmesg:  <http://s000.tinyupload.com/index.php?file_id=97384627797609328044> http://s000.tinyupload.com/index.php?file_id=97384627797609328044

Messages:  <http://s000.tinyupload.com/index.php?file_id=09047399808138966928> http://s000.tinyupload.com/index.php?file_id=09047399808138966928

Motion log: http://s000.tinyupload.com/index.php?file_id=30856176862549134284

 

I have tried several times to upload the motioneye log file, it seems to be too large to upload. It took almost 8 house to download to my computer to send to you. Not sure why it is so large (262,291 kb). Also when I try to connect to MotionPie by SSH I get an error that says “Network Error: Connection Refused” do you know why that is?

Hopefully these other files might help you.

Thanks,

Danny

ccrisan, 2015-09-17:
You lost me here.

1) Did the problem go away by reinstalling?
2) 262kb is not that much - it's a log with many debugging details. Nevermind it as it wouldn't give us more info regarding this issue.
3) The SSH and connection refused problem is a totally different issue than the original one. This usually occurs when the initial boot process hasn't completed successfully (early power cut?).
4) I'm seeing many netcam-related messages in your motion.log. Have you also enabled netcam? The initial reported scenario didn't contain a netcam.

rockman55, 2015-09-17:
Yes the problem seemed to go away after reinstalling. The problem with SSH I think I know what happened to that I can reinstall to get rid of that problem. The netcam related messages, do you mean that my additional cameras  are Network Cameras? If so all the addition cameras are set up as Network Cameras and have always been that way. If there something else you are referring to?

Thanks,

Danny

ccrisan, 2015-09-17:
Well if the problem went away I'll close the issue now. If the problem gets back, please feel free to reopen it.

closed 2015-09-17 by ccrisan