ccrisan/motionpie #296: Aircamdome url for motion pi

bjumy, 2015-09-10:
Hi

i have a raspberry pi 2 and i want to acces an ip camera in the same network 
The rasperry has motionpi and is connected to my network 192.168.30.171
Now the camera is on 30.199 but when i try to add the camera in the motionpi web interface
i can't get it done i need the right url only i do not know which url.

i found that the device url goes like this maybe i am using the wrong port?
rtsp://192.168.30.199:554/live/ch00_0

I need some help i have been at this for ages

thanks

bjumy, 2015-09-10:
it says not a supported network camera but it is a ip network camera so i refuse to believe that

ccrisan, 2015-09-12:
There is no support for RTSP (yet).

closed 2015-09-12 by ccrisan