ccrisan/motionpie #297: 5 mp camera support

USAirconditioning, 2015-09-10:
As 5 MP cameras are available. I would like to modify the resolution to 2560 x1920, but have no idea how to go about it. Any help would be appreciated. Thanks

ccrisan, 2015-09-12:
The resolutions that can be used with a Raspberry PI camera are limited to what shows up in the UI. Such a big resolution would be way too much for the motion daemon running on a PI.

closed 2015-09-12 by ccrisan

USAirconditioning, 2015-09-12:
Thanks for looking into this for me. I found the following  older file  where parameters are adjusted in motion.log.

 

 

http://www.instructables.com/id/Raspberry-Pi-as-low-cost-HD-surveillance-camera/step7/Installing-the-motion-detection-software/

 

They set a width and height and a frame rate.  I had been successful using the 5mp camera to capture still images but was unable to load the additional wireless software. Stuck here:

 

https://learn.adafruit.com/monitor-your-home-with-the-raspberry-pi-b-plus/monitoring-your-home-via-wifi

 

So I opted for the motion pie solution but the video resolution was less than I had hoped for. Not looking for a very high frame rate. Once every 10 seconds would be good enough.

 

Your solution was very easy to implement and almost fit for my purposes.

 

Regards, 

 

Pete

 

This email is free from viruses and malware because avast! Antivirus protection is active.
https://www.avast.com/antivirus