ccrisan/motionpie #298: Weird behavior...

itsnotlevel, 2015-09-12:
Hi Calin, I love MotionPie. I am a Linux user and am very familiar with the Pi (Its a Pi2). I stumbled across your awesome OS last week and here's how everything went. I normally use dd to burn all my images but I liked the fact that with your write image, I could pre-configure the network, making it easier to find my Pi. So, I used writeimage.sh and everything went well. I opened a browser and navigated to the IP of my Pi. Your UI came up instantly and I was very impressed by the time I got the bottom of the page. Very nice, Kudos to you! I was configuring everything, changing things and clicking apply after I changed a thing or two. I got to the end of the configuration menus....clicked apply and a reboot was required...so I did. Pi came back up no issue, refreshed my browser and hmm, something was missing. Most of the menu was no longer there...meaning the things I configured before were no longer there for me to change or see. I was like, huh? I refreshed, turn off advanced menu and back on, apply....nothing. The camera was flipped back over ( I have it mounted upside down) and it was as if I had never configured anything.
I then proceeded to pull the SD, wipe, re-install the OS and put the SD back in the Pi. I fired it up and just like normal, it took a few minutes and the UI was back up again. I proceeded to configure and save and then a reboot....and it was back to square one again! I couldn't believe it. I have re-installed it 5 times and now I am just giving up until I get some answers or else its no good to me really. Sorry for the long email, but I'm trying to give everything. Please let me know as soon as you can. Thank you!

Paul Wilson

ps. I have even run Raspian with Motion on this same Pi before with no issues, just in case you want to know.

ccrisan, 2015-09-12:
A screenshot, please.

itsnotlevel, 2015-09-12:
Sorry....wasn't sure which one you'd respond to. I'll get a screen shot
today! :)

Paul Wilson

drknstrmyknight, 2015-09-14:
I have experienced this when the PI lost network connectivity, and you try to refresh the page.