ccrisan/motionpie #299: Camera rotation not working

markdeblois, 2015-09-12:
Hey guys, I am new to raspberry and motioneye but very interested indeed, seems like a much better alternative to zoneminder which I have used in the past. I am having trouble rotating the cameras, it is ignoring whatever change I make through the UI. Also for resolution. Any quick pointers anyone? I am using an onboard IR camera from raspberry in case that makes a difference.

ccrisan, 2015-09-12:
Can I see a screenshot of your configuration menu (the *Video Device* section, more precisely), as well as the corresponding camera frame? Have you enabled Fast Network Camera?

markdeblois, 2015-09-13:
[image]
Thanks for your quick assistance. Here you go, a screenshot. How do I enable Fast Network Camera? Can't seem to find it...

markdeblois, 2015-09-13:
[image]
This is the corresponding frame. It is too dark now, I need to change the spotlight, you can barely see the grass and waterhole on the bottom right. I need to tilt the camera 90 degrees right, but that is what isn't working.

ccrisan, 2015-09-13:
Nevermind the Fast Network Camera (it's under Expert Settings). If you want 90 degrees, set the video rotation to 90 degrees (it's 180 in your config). Otherwise, I cannot tell anything from the screenshot you sent me. I do believe there's some grass somewhere in the picture though :)

markdeblois, 2015-09-14:
Am I using the same tool, I am not seeing the Expert Settings. I did rotate to 90 degrees and save, my problem is that whatever I pick doesn't reflect. I am using motioneye on a clean raspbian setup, not the motionpie, as I would like to do other stuff with it too.

ccrisan, 2015-09-14:
If you're using motionEye with Raspbian, then please report the problem on [motionEye](https://github.com/ccrisan/motioneye/) rather than here. I assumed that you were using motionPie.

I will try to reproduce the problem on a plain motionEye setup. Please tell me what version of motionEye and what version of motion you are running.

markdeblois, 2015-09-14:
Ah you are right, apologies, I used the wrong place. Not really sure how to check the version, but when I open motion.conf under /etc/motion/ it reads "This config file was generated by motion 3.2.12". I use motioneye version 0.27.2. Should I repost under motioneye?

ccrisan, 2015-09-14:
Let's keep this here for now. The `/etc/motion` directory is not used by motionEye. `/etc/motioneye/` is used instead. If you installed motion using apt-get/aptitude, you're probably running the 3.2.12 version. Otherwise, you could always run `motion -h` and see what version it reports.

markdeblois, 2015-09-14:
I am indeed running 3.2.12, I checked with motion -h. I used the instructions under: https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian.

ccrisan, 2015-09-18:
I'm sorry but I cannot seem to reproduce your problem. Here are a few ideas so that we can carry on with the issue:

1. Please send me a screenshot of the camera frame when the rotation is set to 0 degrees and one from when the rotation is set to 90 degrees. I would be great if the were taken during daylight :)

2. Please describe what happens when you change the rotation angle. Does the orange *Apply* button appear? When you click on it, does your camera frame display a progress animation for a few seconds?