ccrisan/motionpie #301: Problem syncing with NTP

subsong, 2015-09-16:
Hello there,

I have set two Pi's up running the 20150719 release of MotionPie.  They are running in a location without internet access to I set up an NTP server so they can keep time.  The NTP server is a Banana Pro running Raspbian for Banana Pro.  It has a USB GPS adaptor and is successfully syncing its clock from GPS.  When I query it from another machine with ntpdate, I get a response that tells me it's a stratum 1 clock, so it's all looking good.  I cannot get the two MotionPie machines to sync with it though.

Firstly, when I choose NTP, rather than HTTP, the address box in the GUI disappears, so I cannot enter the IP address of my time server.  I managed to overcome that by entering the IP address in the HTTP time server box, then changing from HTTP to NTP.  I then took a backup of the config, which I opened up and examined.  From there, I was able to determine that it was correctly configured to use my NTP server.  Still it is not playing ball though.  I always get 1970 and I am unable to manually update the clock, as this causes a reboot.

I also tried another trick.  I set the config to use HTTP to avoid a conflict, then connected via SSH and tried to use ntpdate to force it.  This works and update the clock, unfortunately the MotionPie software becomes unresponsive and the Pi eventually reboots and comes back up with its clock back in 1970.  I guess the time jump is too massive.

Thanks
Mark

ccrisan, 2015-09-17:
The issue here is the fact that the NTP server is not configurable. The HTTP server field is totally different. This is a low priority issue but a valid one.

*The workaround* is to edit `/etc/ntp.conf` and change the servers there. Never update the system afterwards :)
* The long term solution* would be to implement the same overridding mechanism as for `wpa_supplicant.conf` and others: to use `/data/etc/ntp.conf` along with UI support.

subsong, 2015-09-17:
I tried to edit that file but I was unable to.  I got an error message about a readonly file system.  Is it possible to remove the SD card from the Pi and edit it from another machine, then boot again?
ThanksMark

ccrisan, 2015-09-17:
https://github.com/ccrisan/motionpie/wiki/Tweaks#partition-layout

subsong, 2015-09-17:
Thanks a lot.  I'll remount the partition and make the change as soon as I'm back on site. 

   

 https://github.com/ccrisan/motionpie/wiki/Tweaks#partition-layout—

subsong, 2015-09-23:
I have now made that change and it's all working as it should.  Thanks for your assistance.

ccrisan, 2015-09-23:
The issue will stay open until I have time to make the NTP servers configurable from the UI.

surak, 2015-10-21:
How can I set a timezone? NTP finds UTC, not where I am

surak, 2015-10-21:
The timezone is at the wrong place. Shouldn't it be next to the time server, instead of where it is now?

ccrisan, 2015-10-22:
It's debatable. Timezone is a setting that can and should be altered by all users when configuring their system for the first time. The NTP servers should normally not be altered unless the system runs in some restricted network.

surak, 2015-10-22:
You are right, but perhaps a link to the setting, as one would expect to alter the timezone when setting the time server :-)