ccrisan/motionpie #302: Motioneye on raspberry 2 B+ won't boot

romperstomper, 2015-09-16:
Hi,

I created an sd card with both the 20150719 and the 20150621 release. When I insert the sd card in any of my 3 raspberry pi 2 B+, it stays on the coloured splash screen. I then installed raspbian to the same card and it worked on all 3 raspberry pis.

Unless I am missing something very obvious, I think the build is broken for raspberry pi 2.

willbird, 2015-09-17:
I was running 20150228 on Raspberry Pi 2 B, just got the NOIR camera today, found the 20150228 linked from a question about the Pi 2 B , loaded it up it ran fine. Just put in 20150719 and it too worked fine. 

Is there a Raspberry Pi 2 B+ ?? Do you have B+ or 2 B ?

Bill

ccrisan, 2015-09-17:
I have updated the troubleshooting wiki page to help users rule out a few common mistakes when their systems won't boot: https://github.com/ccrisan/motionpie/wiki/Troubleshooting#system-wont-boot.

Long story short, make sure you have (1) downloaded the correct variant for your PI 2B board (there's no such thing as PI 2B+, yet) and (2) make sure you have extracted the image file prior to writing it to your SD card.

romperstomper, 2015-09-17:
Thank you for responding. I acknowledge board model is PI 2B. Otherwise the
issue remains the same. I have 5 SD cards (2 x 16Gb, 2 x 8Gb, 1 x8Gb). I
have downloaded the correct motionpie variant in the last 2 release dates
(june & july). I have extracted them. I have imaged them onto the cards. I
transfer over the card to each of my 3 boards (all model PI 2B). Please
find attached the screen I see when I attach the power. As a control to
check the health of the process I perform the same steps with Raspbian
(which also comes as a compressed image) and everything works fine. For
completeness I have performed the steps in ubuntu trusty and windows 7. If
you can think of another step/test that I can try in order to track down
the root cause of the issue I am experiencing, I would be happy to try it.

Thank you, gary

I have updated the troubleshooting wiki page to help users rule out a few
common mistakes when their systems won't boot:
https://github.com/ccrisan/motionpie/wiki/Troubleshooting#system-wont-boot.

Long story short, make sure you have (1) downloaded the correct variant for
your PI 2B board (there's no such thing as PI 2B+, yet) and (2) make sure
you have extracted the image file prior to writing it to your SD card.

ccrisan, 2015-09-17:
If you comment via email reply, your attachments get lost. Please attach them here (on github).

romperstomper, 2015-09-18:
best i can do from phone:
https://goo.gl/photos/s2VPprangmQQzZn88

ccrisan, 2015-09-18:
1) Can I see a photo of the board (with the model printed on the PCB)?
2) What is the link from which you have downloaded the image?
3) What are the commands you have used to unpack and write the image? Did you do it from windows or linux (or maybe mac)?

romperstomper, 2015-09-18:
turns out my assumption of which raspi I had was wrong. thanks for helping with this issue, sorry it turned out to be user error!

closed 2015-09-18 by romperstomper