ccrisan/motionpie #304: Multi threading on Pi 2 ?

willbird, 2015-09-17:
Does motionpie use all 4 processor cores on the Pi 2 ?

Sorry if this is an ignorant question :-), when I run Top and press 1 in Rasbian I see the load on all 4 processor cores, when I run Top on motionpie I do not.

Thanks so much for writing motionpie I tried for endless hours to get motion mmal working on Rasbian to no avail, motionpie worked on the first try :-).

Bill

ccrisan, 2015-09-18:
The short answer is yes. The long answer is yes, when needed. That is, if motionPie needs to capture, encode videos, create timelapse videos or zipped pictures in the same time, it will use more than one CPU. If multiple cameras are added to the system, they will also benefit from multiple cores.

If your motionPie does only video capture for one camera, it's very unlikely to see more than one CPU used.

closed 2015-09-18 by ccrisan

willbird, 2015-09-18:
Thank you, if I posted this incorrectly j apologize :-), is there a way to post a question ?

ccrisan, 2015-09-18:
This is the correct and only way to post a question. I answered it and I closed it so that it won't show up as an unsolved issue.

willbird, 2015-09-18:
Ok, new to this, just want to ensure I am not being a PITA :-)