ccrisan/motionpie #305: Motionpie deletes JPGs not related to it on network shares

sv87411, 2015-09-18:
When turning on still image capture and setting the retention period to one month and using a network location for storage, the application removes ALL JPG files from the root of that network location that are older than one month and not just those files related to the camera.

I have confirmed that it is the motionpie app that deleted my files as my monitoring showed that disk space immediately started to increase at the point I turned on this option. Also all files less than a month old were retained. I am very reluctant to re-test to prove.

To replicate.

1) In the  File Storage section. Set the storage device to  option to  "Network Share", enter the details of your share. I chose a standard Samba mount I had on a Linux server and left the Root Directory at root, assuming that the motionpie app would create it's own directory structure.

2) Turn on Still Images capture

3) Set the Still Images Capture Mode to "Interval" and set the Preserve Pictures option to 1 month.

At this point the application starts to remove all pictures from the directory structure that are older than one month. In my case this was every single photo I had ever taken and stored on that share - a lot of data. Luckily I have a backup, but I still lost files that were not backed up. Needless to say I am not impressed.

This is not acceptable behaviour and the application needs to be checked/re-written to only apply it's retention period to it's own files and not others that may reside on the network share!

ccrisan, 2015-09-18:
Yes, this is true. While I agree it's not acceptable in some cases, there's no way the system can distinguish between JPEGs created by it or by other programs.

sv87411, 2015-09-18:
Then I would suggest you at least put a warning in the application telling users that if they point the app at a share with existing jpg/JPG files it will remove EVERY one of them that has a date that falls outside of the retention period. Every one. Not one will remain.

Your app allows a user to innocently setup (as I did) an image grab that saves files to a network share at set intervals that will DELETE their data!

Why can your app not create a single directory called motionpie under the share and store it's image captures under that. Then you only need apply retention policies to jpg/JPG files under that directory structure.

While you may not be prepared to fix the app you should at least provide a warning. I have lost data because of your app and there is a possibility that others will too. You're happy with that?

ccrisan, 2015-09-18:
Am I happy with that? What kind of question is that? I am sorry to hear that you have lost your data but that's not my fault. You're using my work at your own risk and I am in no way responsible of your loss.

Forcing a `motioneye` or a `motionpie` subfolder is not a solution as the users would no longer be able to choose a specific path of their own.

Nevertheless I am going to display a warning regarding the removal of JPEGs in the configured root directory.

sv87411, 2015-09-18:
I should state that I think this software is very good and I was initially very happy with and still am, but you must accept that having lost data my opinion will be tarnished. 

It is a perfectly reasonable question to ask of a developer whose application arbitrarily deletes data not owned or created by it, without any warning to the user. Your first response of "will not fix" assumed that you were perfectly happy that  your application does this and you felt it wasn't necessary to modify it with a warning or permanent fix. I therefore felt absolutely justified to ask whether you were happy with that situation. I certainly wouldn't be.

The application creates directories for each day it creates image capture files and as such is already 'choosing' it's own path to use. If the application created a top level directory called 'motionpie' under the user specified root directory and restricted of all it's writing and deleting activities to that directory structure it would be much safer for the user and not impact on the usability at all. The user still gets to choose a root path, but motionpie then controls a singles directory containing captures below that. This supports your desire for users to "be able to choose a specific path of their own" but simply adds a single directory under it for all of your data. The retention routine can then apply only to that directory instead of the whole of the directory tree from the root path.

I am fully aware that I am using your work at my own risk and I have suffered the consequences of this. I want to ensure that  my experiences are shared and that no one else experience those same consequences. Software that at first looks innocuous, but turns out to be have 'features' that  delete  user data should be highlighted and either fixed or users warned of the consequences of configuring it incorrectly.

I have searched, but found nothing in the Wiki about the file preservation process for still images and that it will delete all jpg/JPG files in the configured root path and below regardless of whether they are motionpie files. So it is necessary to make people aware with a warning message. Preferably the application should store it's files in a single directory structure and only apply retention policies to that structure. If that is too difficult to implement then a warning message stating that all jpg/JPG files older than the retention period chosen will be removed from the root path downwards has to be implemented otherwise there is a risk that more people will lose data.

Thank you.

ccrisan, 2015-09-18:
I agree that my "won't fix" reaction was a bit harsh. I also agree that I need to do something about this and I am really sorry about your loss. I never imagined someone *not* setting a directory specific to motionEye/motionPie when dealing with media files, and that I should have.

As a consequence, the UI will display a big warning when enabling old media file deletion, informing the user about this problem and suggesting to choose a folder specific to motionEye.

drknstrmyknight, 2015-09-18:
MY 2Cents:
Testing any software in a production environment invites disaster.If I gave an application full rights to my important files when there was no necessity for that application to be able to access those files, I would quietly adjust my access control lists and flog myself for a brief period of time.

sv87411, 2015-09-21:
I agree that testing software in a production environment indeed invites disaster. Unfortunately I don't consider my home environment to be in any way a production environment as I suspect a lot of people don't.

I regularly do reckless things on my own PCs and servers that I would never do in a commercial environment. Thus why I have backups of data that are important to me.

Had I tested this software with a dedicated account and share I would never have found this particular peculiarity of the software and thus wouldn't never have raised this bug report and others would never have known that the software has this retention feature that applies to all files on the share. The software makes no indication that it is likely to do this and I think you'll agree that software shouldn't need to be written in this way.

So my mistake and loss is hopefully of benefit to others. Had I taken a test/production approach to my personal IT setup I would never have found this bug and the software would continue to delete less informed people's files with no warning.

Sometimes testing is just a stupid person doing stupid things.

ccrisan, 2015-09-21:
@sv87411 thanks to you the latest version of motionEye has a big warning regarding this behaviour. Again, I am sorry for what happened. I will close the issue now as there's not much else I can do about it.

closed 2015-09-21 by ccrisan

sv87411, 2015-09-21:
Thank you. I am happy with that.