ccrisan/motionpie #307: How to add a motionpie camera to another motionpie instance.

db414813, 2015-09-19:
I have two rpi's. Each running motion pie. I have one with a camera(rpi1) and one without. I want to stream stream the camera(rpi1) from one to my motionpie rpi sever(rpi2). I was able to do a simple mjpeg stream but I want motion recording. So what is the default stream I should use when I try to add a motionpie camera to my server(rpi2)?

ccrisan, 2015-09-20:
As a matter of fact this is how I use motionPie myself: a bunch of RPI1s configured for streaming and a RPI2 with all cameras added as network cameras and with motion detection enabled. You should:

1. Configure your RPI1 to work in Fast Network Camera mode, if and only if you use a RPI CSI camera.
2. On the motionPie that runs on your RPI2 add a network camera with the following URL: http://your_rpi1_ip:8081/.

willbird, 2015-09-20:
So, do you have all the cameras writing files to the same directory ?? That might be kind of neat, as would maybe be the option to start saving images on ALL the cameras when one detects motion :-).

ccrisan, 2015-09-20:
I have all cameras (which are now network cameras) added to one single RPI2 which does the detection and video encoding for all of them. Yes, the media files coming from all cameras are saved in the same root directory. However each camera has its own prefix in this root directory, so I can distinguish between files created by each of them.

db414813, 2015-09-20:
Ok. Do I need to use any authentication like user or admin?

willbird, 2015-09-20:
yes it is nice to have the prefix, but if sorted by creation date/time it could be useful :-).

db414813, 2015-09-20:
I have this for a camera http://www.amazon.com/Waveshare-Raspberry-Camera-Vision-Raspberry-pi/dp/B00N9YWLHE/ref=sr_1_1?ie=UTF8&qid=1442769550&sr=8-1&keywords=Rpi+night+vision+camera. It seems to work well with the fast network camera option on. I used it as a simple mjpeg stream.

ccrisan, 2015-09-20:
@db414813 if you configured your network camera with credentials (the surveillance username doubles as a username for netcam authentication), then yes, you'll need to supply authentication details :)

db414813, 2015-09-20:
OK. So far it's not working. Which log files would you need to see? I would also like to thank you for your help and creating motionpie.

ccrisan, 2015-09-20:
What do you mean it's not working? Do you get an error? Did the add camera dialog work as expected?

db414813, 2015-09-20:
Im out and about now. I believe it said camera not found, I think. I probably didn't set it up right. I will play with it more tonight.

db414813, 2015-09-20:
Use add network camera and not remote motioneye camera correct?

ccrisan, 2015-09-20:
Network camera and *not* remote motionEye camera; correct! Choosing remote motionEye camera would only allow you to remotely control your motionEye camera, which is not what you want.

db414813, 2015-09-21:
I tried it again today and it works!

closed 2015-09-21 by db414813