ccrisan/motionpie #308: Quantization table 0x00 was not defined

Fineburgen, 2015-09-20:
Was giving MotionPie a shot on some older hardware (Raspberry Pi 1 B and a Philips SPC500NC webcam I had lying around) however it seems unable to input from the device, resulting in a gray image within the GUI reporting "Unable to open video device". Upon manually running "motion" results in the below:

http://txt.do/af610

"cat /var/log/motion.log" results in much the same - looped errors of "Quantization table 0x00 was not defined".

Is this the camera sucking too much or something I can fix? I've attempted with the most recent build (20150621) as well as one back (20150719).

ccrisan, 2015-09-20:
I have never seen that error. If this happens with other USB cameras as well then it needs further investigation. If it doesn't, then I'd say it's either a crappy camera or a crappy driver.

donjakobo, 2015-10-15:
I also get this with a Logitech 4000 pro cam. Looks like a driver support issue maybe.

ccrisan, 2015-10-15:
Yes, it looks like a driver problem. Other motion users have reported it: http://www.lavrsen.dk/foswiki/bin/view/Motion/SupportQuestion2011x05x21x132246.

Fineburgen, 2015-10-15:
In my particular case it was the camera that was the problem - can only presume that the Philips thing I had lying around isn't covered by a generic driver. Upon switching to a Logitech C170, everything started working.