ccrisan/motionpie #309: building from source

Bosvark, 2015-09-20:
The image built from source, does not recognise the eth0 interface. I see a eudev error message that says "udev[84]: Could not create /run/udev: No such file or directory". I suspect that is the cause.

What I have noticed, is that udevadm --version gives 219 on your release image (which works btw) and it gives 220 on my image.

ccrisan, 2015-09-20:
The latest commits include a merge with the most recent buildroot. If you wish to build your own version of motionPie, I suggest using one of the tagged commits (e.g. https://github.com/ccrisan/motionpie/tree/20150719). The latest commits haven't been tested too much and things are still in development.

Bosvark, 2015-09-21:
That makes sense, thanks.

Bosvark, 2015-09-22:
I've spent some more time on this and finally got it working. I checked out the tagged version mentioned above, from scratch and did the build. I still got the same problem.

What fixed it, was to remount the rootfs as writeable and to create a symlink from /run to /tmp like so:ln -s /tmp /run. After a reboot, eth0 was recognised and configured properly.

Now, having it working is great, but I would like to understand why the link was not made as part of the build, what am I missing? Among other things, I am also using this project as a learning exercise to understand buildroot better, but I am still confused, even after hours of studying the documentation.

ccrisan, 2015-09-23:
I would also like to know why that happened :) I'll see if I can reproduce the problem myself.

ccrisan, 2015-09-23:
What host OS have you used to build the image? What does `ls -l /path/to/motionpie/board/common/skeleton` say?

Bosvark, 2015-09-24:
I am using ubuntu 14.04. Skeleton directroy looks as follows:
drwxrwxr-x 18 dieter dieter 4096 Sep 22 19:16 .
drwxrwxr-x  4 dieter dieter 4096 Sep 21 09:42 ..
drwxrwxr-x  2 dieter dieter 4096 Sep 21 09:36 bin
drwxrwxr-x  2 dieter dieter 4096 Sep 21 09:36 boot
drwxrwxr-x  2 dieter dieter 4096 Sep 21 09:36 data
drwxrwxr-x  3 dieter dieter 4096 Sep 21 09:36 dev
drwxrwxr-x  4 dieter dieter 4096 Sep 21 09:42 etc
drwxrwxr-x  3 dieter dieter 4096 Sep 21 09:36 home
drwxrwxr-x  2 dieter dieter 4096 Sep 21 09:36 lib
drwxrwxr-x  2 dieter dieter 4096 Sep 21 09:36 media
drwxrwxr-x  2 dieter dieter 4096 Sep 21 09:36 mnt
drwxrwxr-x  2 dieter dieter 4096 Sep 21 09:36 opt
drwxrwxr-x  2 dieter dieter 4096 Sep 21 09:36 proc
drwxrwxr-x  2 dieter dieter 4096 Sep 21 09:36 root
lrwxrwxrwx  1 dieter dieter    3 Sep 22 19:16 run -> tmp
drwxrwxr-x  2 dieter dieter 4096 Sep 21 09:36 sbin
drwxrwxr-x  2 dieter dieter 4096 Sep 21 09:36 sys
drwxrwxr-x  2 dieter dieter 4096 Sep 21 09:36 tmp
drwxrwxr-x  6 dieter dieter 4096 Sep 21 09:36 usr

Also, git status gives:
On branch 20150719
Changes not staged for commit:
  (use "git add/rm <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	deleted:    system/skeleton/var/cache
	deleted:    system/skeleton/var/lib/misc
	deleted:    system/skeleton/var/lib/pcmcia
	deleted:    system/skeleton/var/lock
	deleted:    system/skeleton/var/log
	deleted:    system/skeleton/var/pcmcia
	deleted:    system/skeleton/var/run
	deleted:    system/skeleton/var/spool
	deleted:    system/skeleton/var/tmp

Hope that helps.

ccrisan, 2015-09-25:
Well the skeleton/var directory is somehow missing from your source tree. Can you try to clone the repo in a new directory and see if the problem persists?

Bosvark, 2015-09-29:
Sorry for the delay in getting back to you on this issue, was away. I did a fresh clone of the project and the motionpie/board/common/skeleton/var directory is still missing. In fact, it is also missing in the online view of the source tree, so it is not just on my side where there is a problem.

ccrisan, 2015-09-29:
You were right. The skeleton/var directory was missing from the git tree. This should be fixed with the latest commit.

closed 2015-09-29 by ccrisan