ccrisan/motionpie #311: Can't upload photos

ahollander, 2015-09-20:
After successfully uploading photos, the feature is no longer working, lost a few weeks of recorded pictures. any suggestions.

ccrisan, 2015-09-20:
Please give us more details. Where and how exactly were you uploading your photos? What do you mean the feature is no longer working? Do you get an error? How did you lose your pictures? Did motionPie delete them?

ahollander, 2015-09-20:
The issue is with MotionPie 20150228 on Pi-B

I have already downloaded the newest version and saw no way to recover the time lapse pictures prior.

should have posted earlier.

The message is to the effect "Can not download photos"

I've tried rebooting multiple times, completely shutting it down and powering back up.

tried from chrome, ie, firefox and ois, from three different computers.

I was using only 8% of the available space but tried changing the setting on the camera to a longer time between pictures to see if it might be interfering and also changed the save files from forever to 30 days to see if more space was actually being used and hanging up the program.

no joy.

thanks for any future suggestions.

Aaron

[https://avatars0.githubusercontent.com/u/1835386?v=3&s=400]<https://github.com/ccrisan/motionpie/issues/311#issuecomment-141811128>

Can't upload photos · Issue #311 · ccrisan/motionpie
Read more...<https://github.com/ccrisan/motionpie/issues/311#issuecomment-141811128>

ccrisan, 2015-09-21:
Are you trying to *download* or *upload* photos? Please update the issue description accordingly. If you're trying to download photos, approximately how many photos are you trying to download?

ahollander, 2015-09-21:
When I select the photo icon, I normally get a schedule of pictures by date with the heading "pictures taken by [name of camera]" At that point, I can then select which photos I want downloaded or at the bottom I can then select zipped, timelapse or delete all.  when I selected the photo icon as described above, it would just show the processing icon and either come back with nothing (e.g. as if no request was made) or come back with a message that it couldn't perform the function.  sorry I don't recall the exact message. the schedule above would be about a two days worth of photos to about a week.  I tried this over a number of days with rebooting, restarting, etc.  i didn't want to lose the photos.  It was on a time lapse of 300 seconds.  Note that it was working fine for about a month before it stopped working.
Hope that helps clarify.

ccrisan, 2015-09-21:
That means the number of pictures you're trying to download is too big for your PI to process within a reasonable amount of time. You have two options: (1) upgrade to a more powerful board - e.g. PI2, or (2) try to download a smaller number of pictures. There's nothing else I can do to help you with this.

closed 2015-09-21 by ccrisan