ccrisan/motionpie #312: Update

ahollander, 2015-09-20:
When I check if the software is up to date, it confirms I have the most current, but it isn't. is this a software bug.

ccrisan, 2015-09-20:
Please be more specific. What version do you run and what board do you use? What version do you expect?

ahollander, 2015-09-20:
Calin,

First thank you for getting back so quickly. You've written great software and I've been using it for awhile now.

The issue with not updating the software happens with both the MotionPie 20150228 on Pi-B and on the Pi2 with 20150317.

Hope that clarifies.

Aaron

[https://avatars0.githubusercontent.com/u/1835386?v=3&s=400]<https://github.com/ccrisan/motionpie/issues/312#issuecomment-141811187>

Update · Issue #312 · ccrisan/motionpie
Read more...<https://github.com/ccrisan/motionpie/issues/312#issuecomment-141811187>

ccrisan, 2015-09-21:
The first *stable* version after 20150228 and 20150317 is [20150331](https://github.com/ccrisan/motionpie/releases/tag/20150331). If you read the notes, you'll see that there is an issue with the automatic updates and that you should download and reinstall the latest version manually.

closed 2015-09-21 by ccrisan

ahollander, 2015-09-21:
Thank you Calin!

[https://avatars1.githubusercontent.com/u/76639?v=3&s=400]<https://github.com/ccrisan/motionpie/releases/tag/20150331>

Release 20150331 · ccrisan/motionpie · GitHub
motionpie - A video surveillance OS for single-board computers
Read more...<https://github.com/ccrisan/motionpie/releases/tag/20150331>

-