ccrisan/motionpie #313: MP not detecting motion

TechGiga, 2015-09-21:
I have my camera set up in my bedroom window on the 2nd floor overlooking the street, and it is dark outside. When a car drives by with their headlights on, the camera does not detect it as motion even though I have it set on the lowest frame change threshold. It only took one video of a car stopping, but that one only showed it approaching the stop sign then the video ended. Then during the day, I set the threshold to around 3% and then I get a whole bunch of videos of the leaves on the trees moving. Any idea on how to fix this (especially the first issue)?

TechGiga, 2015-09-24:
It was detecting motion and taking automatic videos at first but not anymore.

TechGiga, 2015-09-27:
I would like to get this working as part of my home surveillance system.

ccrisan, 2015-09-30:
You're the first one to report this issue so the only suggestion I have is to make sure all the motion detection parameters are configured properly. If you don't know the "correct" values for them, just remove and re-add your camera and enable motion movies - the options will have their default values.

mikecarr, 2015-10-13:
I have something similar going on, I have MP in my garage. Whenever the light turns on or motion occurs that is not directly in front of the camera it does not detect it.

TechGiga, 2015-10-13:
The cars passing by just 20 feet directly in front of the camera do not count as motion?

ccrisan, 2015-10-14:
Reduce the motion detection threshold to something below 1% and run some experiments to establish the best value for it. You won't find a perfect value for all possible scenarios, so you'll probably need to fix the threshold so that false positives are detected rather than real motion missed.

TechGiga, 2015-10-14:
It's been set to 0.1% the whole time and all its got is a few false positives and no real motion movies (that show cars moving or squirrels on the lawn for example).

ccrisan, 2015-10-15:
Disable lightswitch detection and see if the problem persists.

TechGiga, 2015-10-15:
It was actually already disabled to begin with. :/

ccrisan, 2015-10-15:
Can you send me your motion.conf and thread-1.conf files? You can download them using the backup button and then replace sensitive data (passwords) with e.g. XXX.

TechGiga, 2015-10-15:
It downloads as a tar.gz file and 7-zip "Can not open file as archive". Also the SFTP structure only has the folders sdcard and storage. sdcard contains pictures and storage is empty.

ccrisan, 2015-10-15:
With SFTP you can access the /data/etc/motion.conf and /data/etc/thread-1.conf files. You need to do some directory changes though.

TechGiga, 2015-10-15:
motion.conf:

# @normal_password XXX
# @enabled on
# @show_advanced on
# @admin_username XXX# @admin_password XXX
# @normal_username XXX

thread thread-1.conf
webcontrol_port 7999

thread-1.conf:

# @working_schedule_type during
# @webcam_resolution 100
# @enabled on
# @name RPi Camera Module
# @network_server 
# @id 1
# @motion_detection on
# @network_username 
# @storage_device custom-path
# @preserve_movies 365
# @preserve_pictures 365
# @network_share_name 
# @working_schedule -|-|-|-|-|-|-
# @webcam_server_resize off
# @network_password 

lightswitch 0
snapshot_interval 30
minimum_motion_frames 1
stream_motion on
target_dir /data/output
text_double on
height 480
stream_auth_method 0
stream_quality 85
noise_tune on
threshold 307
quality 80
post_capture 4
noise_level 31
framerate 24
text_left rpisurv_mp-83ac3671
pre_capture 5
width 640
picture_filename 
locate_motion_style redbox
locate_motion_mode off
contrast 0
stream_maxrate 24
output_pictures off
hue 0
saturation 0
stream_localhost off
videodevice /dev/video0
ffmpeg_variable_bitrate 0
ffmpeg_video_codec msmpeg4
text_changes off
movie_filename %Y-%m-%d/%H-%M-%S
auto_brightness off
on_event_end /programs/motioneye/eventrelay.py stop %t
stream_por

ccrisan, 2015-10-16:
Do you by any chance have the Working Schedule option enabled while all the day suboptions are off?

TechGiga, 2015-10-16:
Aha! That's it. Yes, the Working Schedule option was enabled and all the day suboptions are off. So now motion detection will work if I turn off Working Schedule altogether?

TechGiga, 2015-10-16:
I set the motion detection threshold to 2%, and sometimes the camera frame on the webpage is red and the stream is laggy. Why is this?

TechGiga, 2015-10-16:
Oh lol! It's already taken 7 motion movies. Gotta adjust it to be less sensitive. Issue solved. :+1:

closed 2015-10-16 by TechGiga

mikecarr, 2015-10-16:
​Mine is disabled.

ccrisan, 2015-10-16:
1) I made a small change so that the working schedule cannot be enabled unless enabling at least one day. This will prevent situations like this one.
2) The red frame motion indicator is known to be laggy, especially on RPi1. There's not much I can do about it.
3) @mikecarr, how about the lightswitch? If your lightswitch detection is disabled and you cannot manage to find a good motion detection threshold, then you'll need to wait for a future version that will include a mask for selecting the detection area.

mikecarr, 2015-10-16:
Images and movies are being created, just no email. I also do not see any
mention of email being sent or attempted.

https://dl.dropboxusercontent.com/u/17677610/motioneye-config.zip

https://dl.dropboxusercontent.com/u/17677610/Screen%20Shot%202015-10-16%20at%209.14.37%20AM.png

mikecarr, 2015-10-17:
I just loaded it up on a RaspberryPi2 and I am having the same issue,
motions detected no email sent. I have also tried non-tls with the same
issue. I see the email settings in thread-1.conf

on_event_start /usr/bin/python
/usr/lib/python2.7/site-packages/motioneye/meyectl.pyc relayevent -c
/data/etc/motioneye.conf -l start %t; /usr/bin/python
/usr/lib/python2.7/site-packages/motioneye/meyectl.pyc sendmail -c
/data/etc/motioneye.conf -l 'smtp.gmail.com' '465' 'mcarr67@gmail.com'
'xxxxxx' 'False' 'mcarr67@gmail.com' 'motion_start' '%t'
'%Y-%m-%dT%H:%M:%S' '10'

2015-10-16 22:49:26: [sendmail]    DEBUG: hello!
2015-10-16 22:49:26: [sendmail]    DEBUG: server = smtp.gmail.com
2015-10-16 22:49:26: [sendmail]    DEBUG: port = 465
2015-10-16 22:49:26: [sendmail]    DEBUG: account = mcarr67@gmail.com
2015-10-16 22:49:26: [sendmail]    DEBUG: password = ******
2015-10-16 22:49:26: [sendmail]    DEBUG: server = smtp.gmail.com
2015-10-16 22:49:26: [sendmail]    DEBUG: tls = false
2015-10-16 22:49:26: [sendmail]    DEBUG: to = mcarr67@gmail.com
2015-10-16 22:49:26: [sendmail]    DEBUG: msg_id = motion_start
2015-10-16 22:49:26: [sendmail]    DEBUG: camera_id = 1
2015-10-16 22:49:26: [sendmail]    DEBUG: moment = 2015-10-16 22:49:24
2015-10-16 22:49:26: [sendmail]    DEBUG: smtp timeout = 60
2015-10-16 22:49:26: [sendmail]    DEBUG: timespan = 10
2015-10-16 22:49:26: [sendmail]    DEBUG: reading camera config from
/data/etc/thread-1.conf...
2015-10-16 22:49:26: [sendmail]    DEBUG: reading main config from file
/data/etc/motion.conf...
2015-10-16 22:49:26: [sendmail]    DEBUG: loading additional config
structure for main, without separators
2015-10-16 22:49:26: [sendmail]    DEBUG: additional config section: network
2015-10-16 22:49:26: [sendmail]    DEBUG: additional config section:
services
2015-10-16 22:49:26: [sendmail]    DEBUG: additional config section:
expertSettings
2015-10-16 22:49:26: [sendmail]    DEBUG: additional config item:
wifiEnabled
2015-10-16 22:49:26: [sendmail]    DEBUG: additional config item:
wifiNetworkName
2015-10-16 22:49:26: [sendmail]    DEBUG: additional config item:
wifiNetworkKey
2015-10-16 22:49:26: [sendmail]    DEBUG: additional config item: timeZone
2015-10-16 22:49:26: [sendmail]    DEBUG: additional config item:
ipConfigType
2015-10-16 22:49:26: [sendmail]    DEBUG: additional config item:
ipConfigStaticAddr
2015-10-16 22:49:26: [sendmail]    DEBUG: additional config item:
ipConfigStaticMask
2015-10-16 22:49:26: [sendmail]    DEBUG: additional config item:
ipConfigStaticGw
2015-10-16 22:49:26: [sendmail]    DEBUG: additional config item:
ipConfigStaticDns
2015-10-16 22:49:26: [sendmail]    DEBUG: additional config item: ftpEnabled
2015-10-16 22:49:26: [sendmail]    DEBUG: additional config item: ftpAuth
2015-10-16 22:49:26: [sendmail]    DEBUG: additional config item:
ftpWritable
2015-10-16 22:49:26: [sendmail]    DEBUG: additional config item: smbEnabled
2015-10-16 22:49:26: [sendmail]    DEBUG: additional config item: smbAuth
2015-10-16 22:49:26: [sendmail]    DEBUG: additional config item:
smbWritable
2015-10-16 22:49:26: [sendmail]    DEBUG: additional config item: sshEnabled
2015-10-16 22:49:26: [sendmail]    DEBUG: additional config item: watchLink
2015-10-16 22:49:26: [sendmail]    DEBUG: additional config item:
watchLinkTimeout
2015-10-16 22:49:26: [sendmail]    DEBUG: additional config item:
watchConnect
2015-10-16 22:49:26: [sendmail]    DEBUG: additional config item:
watchConnectHost
2015-10-16 22:49:26: [sendmail]    DEBUG: additional config item:
[... 130 lines cut ...]
2015-10-16 22:50:07: [motioneye]    DEBUG: running garbage collector for
mjpg clients...
2015-10-16 22:50:10: [motioneye]    DEBUG: 200 GET
/static/img/motioneye-logo.svg (127.0.0.1) 5.19ms
2015-10-16 22:50:15: [motioneye]    DEBUG: 200 GET
/static/img/motioneye-logo.svg (127.0.0.1) 5.25ms
2015-10-16 22:50:16: [sendmail]    DEBUG: running garbage collector for
mjpg clients...
2015-10-16 22:50:17: [motioneye]    DEBUG: checking SMB mounts...
2015-10-16 22:50:17: [motioneye]    DEBUG: listing smb mounts...
2015-10-16 22:50:17: [motioneye]    DEBUG: running garbage collector for
mjpg clients...
2015-10-16 22:50:20: [motioneye]    DEBUG: 200 GET
/static/img/motioneye-logo.svg (127.0.0.1) 4.94ms
2015-10-16 22:50:25: [motioneye]    DEBUG: 200 GET
/static/img/motioneye-logo.svg (127.0.0.1) 4.89ms
2015-10-16 22:50:26: [sendmail]    DEBUG: running garbage collector for
mjpg clients...
2015-10-16 22:50:27: [motioneye]    DEBUG: running garbage collector for
mjpg clients...
2015-10-16 22:50:30: [motioneye]    DEBUG: 200 GET
/static/img/motioneye-logo.svg (127.0.0.1) 5.23ms
2015-10-16 22:50:35: [motioneye]    DEBUG: 200 GET
/static/img/motioneye-logo.svg (127.0.0.1) 5.15ms
2015-10-16 22:50:36: [sendmail]    DEBUG: bye!

ccrisan, 2015-10-17:
Yeap. Many users have reported email issues. I'll try to fix the problem asap.

mikecarr, 2015-10-17:
Of you need any logs or testing let me know.

ccrisan, 2015-10-20:
The email issue has been fixed upstream in motionEye with version 0.28.2. It will soon be included in the next motionEyeOS version.

mikecarr, 2015-10-20:
Can we manually upgrade?

ccrisan, 2015-10-20:
Nope. Hold on a couple of days and i'll release the new os image.

mikecarr, 2015-10-20:
Thanks for the quick turn around.

ccrisan, 2015-10-22:
20151021 is out. Please test it and let me know if the problem persists.

mikecarr, 2015-10-23:
Working great for me, thanks again for the quick turn around.

mikecarr, 2015-10-23:
Sorry all if this is not the right place but I could have sworn there was a
way to change the picture size, for some reason I cannot find that setting.

ccrisan, 2015-10-23:
Are you talking about the video device resolution? Do you use a network camera? Or have you physically changed the camera device without re-adding it to motionEye? Perhaps a screenshot would help us understand what the problem is.

mikecarr, 2015-10-23:
No problem, is it possible to change to image size? I have raspberry pi
camera module.

ccrisan, 2015-10-23:
You haven't answered my 3 questions above.

mikecarr, 2015-10-23:
I figured it was a yes or no question, it either supports it or doesn't.

Are you talking about the video device resolution? No, still image
resolution
Do you use a network camera? No, Raspberry Pi Camera Module
Or have you physically changed the camera device without re-adding it to
motionEye? No

(ignore the no picture image, this normally works when I am connected)

[image: Inline image 1]
Perhaps a screenshot would help us understand what the problem is.

ccrisan, 2015-10-23:
I'm sorry I am not able to see the image you sent. The still images will always be saved at the exact same resolution at which the video device works. This never was configurable (at least not with motionEye).

mikecarr, 2015-10-23:
Ok, thank you.