ccrisan/motionpie #316: mjpg client connection problem.

vdyo, 2015-09-22:
Hi

I put a message in before about rtsp streams, I worked out a HTTP stream which is http://10.0.0.51/snapshot.cgi?chan=1 it seems to work when I first add it then it stops. I see these errors from the logs.

2015-09-23 00:00:27: WARNING: Connect error on fd 12: ECONNREFUSED
2015-09-23 00:00:27: WARNING: Connect error on fd 13: ECONNREFUSED
2015-09-23 00:00:27: WARNING: Connect error on fd 14: ECONNREFUSED
2015-09-23 00:00:27: ERROR: connection problem detected for mjpg client for camera 2 on port 8082
2015-09-23 00:00:32: ERROR: connection problem detected for mjpg client for camera 1 on port 8081
2015-09-23 00:00:32: ERROR: connection problem detected for mjpg client for camera 3 on port 8083

ccrisan, 2015-09-22:
There are many known HTTP netcam bugs that have been fixed in the meantime and will be released with the next version.

vdyo, 2015-09-22:
Hi,

When is the next version being released? Can I get a pre release?

Damian

vdyo, 2015-10-06:
Hi

Any idea on the next version release, I am very keen to have a go with this software after using Sighthound for years. 

Regards

Damian

ccrisan, 2015-10-06:
There will hopefully be a prerelease available within 1-2 weeks.

vdyo, 2015-10-06:
Ok Cool, I am happy to test.

vdyo, 2015-10-12:
Ok, so I should be running motioneyeos now? Is that the fix?

Thanks

Damian

ccrisan, 2015-10-12:
Yes, the fixes that I've been talking about and that will hopefully solve your problem are available now with motionEyeOS. Please test it and get back to me with results. Many other underlying changes have been made and other things might have gotten broken. That's why it's just a prerelease, for now.

vdyo, 2015-10-13:
Hi, Ok I have re imaged my Pi, the cameras now work well, have been tweaking the motion detection, so thats working well. I cant seem to get the email notifications working, it worked once but not again, I am not sure why. A "Test Email" function would be great, so you know at least that is working. The other thought is a tick to remember me on the user survelance mode, I added a link on my iPhone but you need to log in each time which is a pain. In saying all that well done, its working great.

ccrisan, 2015-10-13:
Taking a look over `motion.log` may give us a hint regarding why email is not working. I agree that a test button would be great.

vdyo, 2015-10-14:
Hi, Ok the Log is 

```[0] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-1.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-2.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-3.conf
[0] [NTC] [ALL] motion_startup: Using default log level (NTC) (6)
[0] [NTC] [ALL] motion_startup: Motion Unofficial-Git-a5b5f13 Started
[0] [NTC] [ALL] motion_startup: Logging to syslog
[0] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (NTC)
[0] [NTC] [ENC] ffmpeg_init: ffmpeg LIBAVCODEC_BUILD 3680612 LIBAVFORMAT_BUILD 3679332
[0] [NTC] [ALL] main: Thread 1 is from thread-1.conf
[0] [NTC] [ALL] main: Thread 1 is device: rtsp://10.0.0.51:554/live/ch00_0 input -1
[0] [NTC] [ALL] main: Stream port 8081
[0] [NTC] [ALL] main: Thread 2 is from thread-2.conf
[1] [NTC] [ALL] motion_init: Thread 1 started , motion detection Enabled
[0] [NTC] [ALL] main: Thread 2 is device: rtsp://10.0.0.52:554/live/ch00_0 input -1
[1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://10.0.0.51:554/live/ch00_0)
[0] [NTC] [ALL] main: Stream port 8082
[0] [NTC] [ALL] main: Thread 3 is from thread-3.conf
[2] [NTC] [ALL] motion_init: Thread 2 started , motion detection Enabled
[0] [NTC] [ALL] main: Thread 3 is device: rtsp://10.0.0.53:554/live/ch00_0 input -1
[2] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://10.0.0.52:554/live/ch00_0)
[0] [NTC] [ALL] main: Stream port 8083
[0] [NTC] [ALL] main: Waiting for threads to finish, pid: 30445
[3] [NTC] [ALL] motion_init: Thread 3 started , motion detection Enabled
[3] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://10.0.0.53:554/live/ch00_0)
[0] [NTC] [STR] httpd_run: motion-httpd testing : IPV4 addr: 127.0.0.1 port: 7999
[0] [NTC] [STR] httpd_run: motion-httpd Bound : IPV4 addr: 127.0.0.1 port: 7999
[0] [NTC] [STR] httpd_run: motion-httpd/Unofficial-Git-a5b5f13 running, accepting connections
[0] [NTC] [STR] httpd_run: motion-httpd: waiting for data on 127.0.0.1 port TCP 7999
[1] [NTC] [NET] netcam_rtsp_open_context: Using tcp transport
[3] [NTC] [NET] netcam_rtsp_open_context: Using tcp transport
[2] [NTC] [NET] netcam_rtsp_open_context: Using tcp transport
[2] [NTC] [NET] netcam_connect_rtsp: Camera connected
[2] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 1 items
[2] [ALR] [NET] netcam_handler_loop: Camera handler thread [4] started
[2] [NTC] [STR] http_bindsock: motion-stream testing : IPV4 addr: 0.0.0.0 port: 8082
[2] [NTC] [STR] http_bindsock: motion-stream Bound : IPV4 addr: 0.0.0.0 port: 8082
[2] [NTC] [ALL] motion_init: Started motion-stream server in port 8082 auth Disabled
[2] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 3 items
[1] [NTC] [NET] netcam_connect_rtsp: Camera connected
[1] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 1 items
[1] [ALR] [NET] netcam_handler_loop: Camera handler thread [5] started
[1] [NTC] [STR] http_bindsock: motion-stream testing : IPV4 addr: 0.0.0.0 port: 8081
[1] [NTC] [STR] http_bindsock: motion-stream Bound : IPV4 addr: 0.0.0.0 port: 8081
[1] [NTC] [ALL] motion_init: Started motion-stream server in port 8081 auth Disabled
[1] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 3 items
[1] [NTC] [EVT] event_new_video FPS 3
[1] [NTC] [ENC] ffmpeg_open Selected Output FPS 3
[1] [NTC] [EVT] event_newfile: File of type 8 saved to: /data/2015-10-14/01-28-29.avi
[1] [NTC] [ALL] motion_detected: Motion detected - starting event 1
[1] [NTC] [EVT] event_newfile: File of type 1 saved to: /data/2015-10-14/01-28-29.jpg
[1] [NTC] [EVT] event_newfile: File of type 1 saved to: /data/2015-10-14/01-28-29.jpg
[3] [NTC] [NET] netcam_connect_rtsp: Camera connected
[3] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 1 items
[3] [ALR] [NET] netcam_handler_loop: Camera handler thread [6] started
[1] [NTC] [EVT] event_newfile: File of type 1 saved to: /data/2015-10-14/01-28-29.jpg
[3] [NTC] [STR] http_bindsock: motion-stream testing : IPV4 addr: 0.0.0.0 port: 8083
[3] [NTC] [STR] http_bindsock: motion-stream Bound : IPV4 addr: 0.0.0.0 port: 8083
[3] [NTC] [ALL] motion_init: Started motion-stream server in port 8083 auth Disabled
[3] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 3 items
[1] [NTC] [EVT] event_newfile: File of type 1 saved to: /data/2015-10-14/01-28-30.jpg
[1] [NTC] [EVT] event_newfile: File of type 1 saved to: /data/2015-10-14/01-28-31.jpg
[1] [NTC] [EVT] event_newfile: File of type 1 saved to: /data/2015-10-14/01-28-32.jpg
[1] [NTC] [EVT] event_newfile: File of type 1 saved to: /data/2015-10-14/01-28-33.jpg
[1] [NTC] [EVT] event_newfile: File of type 1 saved to: /data/2015-10-14/01-28-35.jpg
[1] [NTC] [EVT] event_newfile: File of type 1 saved to: /data/2015-10-14/01-28-36.jpg
[3] [NTC] [EVT] event_new_video FPS 4
[... 106 lines cut ...]
[1] [NTC] [EVT] event_newfile: File of type 1 saved to: /data/2015-10-14/01-31-21.jpg
[3] [NTC] [EVT] event_newfile: File of type 1 saved to: /data/2015-10-14/01-31-21.jpg
[1] [NTC] [EVT] event_newfile: File of type 1 saved to: /data/2015-10-14/01-31-23.jpg
[1] [NTC] [EVT] event_newfile: File of type 1 saved to: /data/2015-10-14/01-31-24.jpg
[3] [NTC] [EVT] event_newfile: File of type 1 saved to: /data/2015-10-14/01-31-24.jpg
[... cut ...]
[3] [NTC] [EVT] event_newfile: File of type 1 saved to: /data/2015-10-14/01-31-51.jpg
[3] [NTC] [EVT] event_newfile: File of type 1 saved to: /data/2015-10-14/01-31-54.jpg
[3] [NTC] [EVT] event_newfile: File of type 1 saved to: /data/2015-10-14/01-31-57.jpg
[3] [NTC] [EVT] event_newfile: File of type 1 saved to: /data/2015-10-14/01-32-00.jpg
[3] [NTC] [EVT] event_newfile: File of type 1 saved to: /data/2015-10-14/01-32-02.jpg
[3] [NTC] [EVT] event_newfile: File of type 1 saved to: /data/2015-10-14/01-32-05.jpg
 ```

ccrisan, 2015-10-14:
You're the second user to report email notification problems with the new image. I'll have to dig deeper into this as your log doesn't indicate anything wrong.

vdyo, 2015-10-14:
I tried to send a new motion.log, but its empty now for some reason.

ccrisan, 2015-10-15:
Ah, nevermind. You could send me the thread-1.conf file instead, **replacing your passwords and sensitive data with, say, xxx**.

vdyo, 2015-10-15:
That might be above my linux know how...where do I find that file, I will
give it a crack.

ccrisan, 2015-10-15:
Backup your configuration using the backup button in the settings panel, unpack the archive and you'll find the thread-1.conf file.

vdyo, 2015-10-15:
Ok that was easy, its attached.

ccrisan, 2015-10-15:
Nope it's not. If you replied to the email received via github, the attachment was ignored. Please send it to me directly at ccrisan around gmail.

closed 2015-10-20 by ccrisan