ccrisan/motionpie #318: run a command info

pengocha, 2015-09-24:
Hi all, i was wondering i could get some more info about the 'run a command' function i found on motion detection considering i'm pretty new to all this. How exactly do i use it? Say for example i want to run a python script should i just type 'sudo python /path/to/script.py' or something? Thank you so much for the wonderfull program !

ccrisan, 2015-09-25:
There's no `sudo` in motionPie - you're always `root`. If you have a python script that you would like to run, you should place it in `/data` (e.g. `/data/yourscript.py`), make it executable (i.e. `chmod +x /data/yourscript.py`) and simply add the path to the *Run A Command* UI entry (i.e. `/data/yourscript.py`).

closed 2015-09-25 by ccrisan

pengocha, 2015-09-25:
Ok, tried it with a basic led on script, heres what i did:
1) ssh into motionpie
2) $ nano /data/testled.py
3) code testled.py=
import RPi.GPIO as GPIO
import time
siren = 17
GPIO.setmode(GPIO.BCM)
GPIO.setup(siren, GPIO.OUT)
GPIO.output(siren,GPIO.HIGH)
time.sleep(100)

[image]

ccrisan, 2015-09-25:
There's an issue with the GPIO and RPI2 (yet to be fixed). Nevertheless, check the motion.log - it should contain relevant output if some errors occurred during the execution of your script.

pengocha, 2015-09-25:
Looks like certain commands aren't found, could that be distro related? (i'm pretty new to linux :s)

/data/testled.py: line 2: import: command not found
/data/testled.py: line 4: siren: command not found
/data/testled.py: line 6: syntax error near unexpected token `GPIO.BCM'
/data/testled.py: line 6: `GPIO.setmode(GPIO.BCM)'

ccrisan, 2015-09-25:
Add `#!/usr/bin/python` at the top of your script.

pengocha, 2015-09-25:
That did the trick, many thanks ccrisan !!