ccrisan/motionpie #319: Email notification - seperate username and email sender address

rkreinz, 2015-09-25:
Hi ccrisan, 

First off all, thank you for this great piece of software!

I would like to see a separate user / email field for the smtp settings in the future, 
because some providers requesting username/password instead of email/pass for authentication.
When I tried to use a given username/password combo for my smtp server, 
motioneye is generating an error message likes this:

SMTPSenderRefused: (553, '5.5.4 <user1>... Domain name required for sender address user1', 'user1')

It works flawless with an gmail account, including the domain name. 
Thank you!

Robert

ccrisan, 2015-09-26:
I understand your problem. I will consider adding support for separate sender field.

ccrisan, 2015-09-27:
How about using the destination email address as the *To* field as well (instead of the *SMTP Account*)? Would that work for you? I'm trying to avoid adding another option for the email configuration; the existing options are already confusing for some users :)

ccrisan, 2015-09-28:
From now on the *From* field is filled in with the first destination address.

closed 2015-09-28 by ccrisan