ccrisan/motionpie #321: Onboard wifi support for Bananna Pi Pro ??

willbird, 2015-09-26:
The board has onboard AP6181 WiFi module. Wifi does not seem to work.

ccrisan, 2015-09-26:
I only have the original Banana PI board. What is the kernel module that needs to be enabled for this wifi module to work?

willbird, 2015-09-27:
I did confirm that it DOES work in rasbian for bananna pi.

ap6210 is the module

I shall start a new issue for SATA on the same board

ccrisan, 2015-09-27:
The ap6210 module does not seem to be included in the mainline kernel (as of 4.2). In order for me to include this driver I would need to either downgrade to the 3.18 kernel used by the "official" Banana PI distros or to include and maintain an out-of-tree driver for the current kernel version. Both variants are bad and I'm not willing to pursue any of them. Let's wait for the lemaker team to mainline the wifi driver.

closed 2015-09-27 by ccrisan

willbird, 2015-09-27:
TY for looking at it, wifi dongle is the solution for motionpie for present then :-).