ccrisan/motionpie #322: help with settings to receive email notifications

suparit1, 2015-09-26:
good evening Mr Crisan.  Hope you're doing great.  Love your motionPie.  Just cannot figure how to set to receive email when motion is detected.

still images is on
motion movies is off

email notifications is on
email address is christopus@gmail.com
smtp server is smtp.gmail.com
smtp port is 587
smtp account is christopus@gmail.com
smtp password is MY GMAIL PASSWORD
use tls is on
attached pictures time span is 5

Do I need to configure any settings in my gmail account?

Thank you.

Chris S

ccrisan, 2015-09-26:
Please send me the motion.log file (downloadable from the *Expert Settings* section).

jsmacdou, 2015-09-28:
There is a setting somewhere in Gmail you need to set, to allow your address to work with external apps.

suparit1, 2015-09-29:
Do u think Mr Crisan that MotionPie goes to sleep after periods of inactivity?  Is there a setting to have it remain active indefinitely?

Chris S

ccrisan, 2015-10-04:
Nope, there's no such setting.

closed 2015-10-04 by ccrisan

suparit1, 2015-10-06:
Thank you.  Able to now receive e-mail notifications.  Thinking it would be
cool if there's a way to limit the area that motionPie detects.

eg.  to block out leaves on the tree that are constantly moving... just a
thought...

Castor66, 2015-12-12:
Hello,

I'm having a similar problem.  I have my Gmail settings set up exactly like the original poster did above, but for my own Gmail account obviously.  But it seems like Gmail blocked my email from being sent, as I got a "suspicious activity email from Gmail.  Someone referenced above a setting in Gmail I can set to allow access from outside apps, but I don't know where that is or what to do.  Any help here would be great!

Thanks,
Josh

jsmacdou, 2015-12-12:
You'll find the relevant setting at https://myaccount.google.com/ under the "Connected apps & sites" section. At the bottom of this section, make sure "Allow less secure apps" is set to ON.

Castor66, 2015-12-12:
That was it - thanks so much!!!