ccrisan/motionpie #324: Sata on Banana Pi Pro

willbird, 2015-09-27:
The version I ran to confrim wifi and SATA do in fact work...
Linux lemaker 3.4.103 #4 SMP PREEMPT Thu Dec 18 12:55:58 CST 2014 armv7l GNU/Linux

The board does appear to run motionpie just fine, I can boot it up Port forward and let you in with some help ccrisan if you want to poke around in a Pro that is running....I have only run it with the following camera.

http://www.newark.com/embest/cam8200-u-module/camera-module-usb-interface-5mp/dp/83X8624

To make SATA work on this board  in Rasbian for BananAPro i went into /etc/modules and uncommented this line....

sw_ahci_platform

# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.

#For SATA Support
sw_ahci_platform

#Display and GPU
#lcd
#hdmi
ump
#disp
mali
mali_drm
ap6210

willbird, 2015-09-28:
/dev/sda1               916.8G      2.3G    867.8G   0% /data/media/sda1

Looks like sata worked out aok :-).

ccrisan, 2015-09-29:
So is this motionPie or some other OS? Are we talking about Banana PI or Banana PRO?

willbird, 2015-09-29:
I am sorry to be obscure :-). Banana Pi Pro. Checked to be sure Sata in general worked by running...

Linux lemaker 3.4.103 #4 SMP PREEMPT Thu Dec 18 12:55:58 CST 2014 armv7l GNU/Linux

Then ran motionpie-bananapi-20150719.img.gz and Sata works with that as well. I suppose I should not be surprised that it worked ;-).

ccrisan, 2015-09-29:
I will close the issue then.

closed 2015-09-29 by ccrisan

willbird, 2015-09-29:
yes :-), I just thought you might like to know it works great :-).