ccrisan/motionpie #325: An error occured. Refreshing is recommended. AFTER rebooting

TechGiga, 2015-09-29:
I cannot access any pictures or motion videos from the motionpie site anymore. It always says An error occured. Refreshing is recommended. Only 1.1 GB of the SD card has been used up out of 30. Please help!

ccrisan, 2015-09-29:
Most likely there are too many media files for the system and it runs out of memory or it timeouts. Please send me the `motioneye.log` file to find out what the problem is.

TechGiga, 2015-09-29:
http://www.gigawertblog.byethost17.com/motioneye.log

ccrisan, 2015-09-29:
You seem to experience the out of memory bug. It's a known bug still under investigation. A reboot of the unit is the workaround for now. If the error persists after reboot, you need to remove some pictures/videos. The media browser cannot cope with too many media files on a Raspberry PI.

TechGiga, 2015-09-29:
OK, then. Fortunately I can still access the pictures via SFTP.

Another somewhat odd issue is this: https://github.com/ccrisan/motionpie/issues/313

drknstrmyknight, 2015-09-29:
I started saving to a windows machine and not trying to use the built-in viewer.I only try to view the files from a desktop/

donjakobo, 2015-10-15:
The only "real" solution I have found for out of memory is physically restarting the RPI, doing a motionpie "reboot" still results in a hung RPI, no video working, or images being captured.  Motionpie starts up tho and is web accessible.

TechGiga, 2015-10-15:
No, you actually just need to download the pictures off the Pi and delete them.