ccrisan/motionpie #328: Is there any roadmap available?

mastix, 2015-10-05:
Hey guys,

first of all, I love your project! Pretty cool idea!

There's one thing I'm missing though... a roadmap. Just to see when new releases are being published.

Do you have any plans to add one?

Greetz,

Sascha

ccrisan, 2015-10-05:
For now there's a [roadmap for motionEye](https://github.com/ccrisan/motioneye/wiki/Roadmap) but I've been meaning to maintain one for motionPie (the future *motionEyeOS*) as well, as the existing list also contains tasks that belong with motionPie rather than with motionEye.

BaGRoS, 2015-10-06:
h264 movie encoding
+++
Using GPU... Raspberry PI & 2 :)

snagglewest, 2015-10-07:
Might you add #232 to the short list? 
https://github.com/ccrisan/motionpie/issues/232

ccrisan, 2015-10-07:
It's number 5 on the list.