ccrisan/motionpie #329: Network share - gone

BaGRoS, 2015-10-06:
In "File Storage" I used
'Network share' :
Network Server  192.168.1.222
Share Name  cam
Share Username  USER
Share Password  •••••••••••
Root Directory /c1
Disk Usage  0.0/0.0 GB (0%)
my samba server //192.168.1.222/cam/c1
After APPLY everything is ok, a few dozen minutes, and then the resource disappears.
I installed the fstab
//192.168.1.222/cam /media/cam cifs username=USER,password=PASSWD,iocharset=utf8,sec=ntlm        0       0
Unfortunately, after the restart it will not mount automatically, by what I have to do this command:
mount /media/cam
after that everything starts to work on 
Custom Path
/media/cam/c1

What You need? Log?

ccrisan, 2015-10-06:
You're not supposed to manually edit your fstab file. Please send me the motioneye.log file.

BaGRoS, 2015-10-06:
How?
I send by email.

ccrisan, 2015-10-06:
https://github.com/ccrisan/motionpie/wiki/Reporting-An-Issue#attach-log-files.

BaGRoS, 2015-10-06:
http://s000.tinyupload.com/index.php?file_id=71155973952906680540

ccrisan, 2015-10-07:
Please remove the fstab line you added, make sure your network share settings are correct (including username and password), then reboot. Let's see if the problem persists.

BaGRoS, 2015-10-07:
[image]
[image]

2015-10-07 13:43:28: ERROR: connection problem detected for mjpg client for camera 1 on port 8765
2015-10-07 13:43:32: ERROR: connection problem detected for mjpg client for camera 1 on port 8765
2015-10-07 13:49:37: INFO: interrupt signal received, shutting down...
2015-10-07 13:49:37: INFO: server stopped
2015-10-07 13:49:37: INFO: interrupt signal received, shutting down...
2015-10-07 13:49:51: INFO: hello! this is motionPie 20150719
mount error(13): Permission denied
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
mount error(13): Permission denied
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
2015-10-07 13:49:54: INFO: cleanup started
2015-10-07 13:49:54: INFO: wsswitch started
2015-10-07 13:49:54: INFO: thumbnailer started
2015-10-07 13:49:55: INFO: server started
2015-10-07 13:51:36: WARNING: control hue not found for device /dev/video0
2015-10-07 13:51:36: WARNING: control hue not found for device /dev/video0
2015-10-07 13:51:37: WARNING: control hue not found for device /dev/video0
mount error(13): Permission denied
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
mount error(13): Permission denied
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
mount error(16): Device or resource busy
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
2015-10-07 13:54:51: ERROR: failed to mount smb share "//192.168.1.222/cam" at "/data/media/motioneye_192_168_1_222_cam_bagros"
umount: can't umount /data/media/motioneye_192_168_1_222_cam: No such file or directory
2015-10-07 13:54:51: ERROR: failed to unmount smb share "//192.168.1.222/cam" from "/data/media/motioneye_192_168_1_222_cam"
2015-10-07 13:55:12: WARNING: control hue not found for device /dev/video0
2015-10-07 13:55:12: ERROR: failed to execute statvfs: [Errno 95] Operation not supported: '/data/media/motioneye_192_168_1_222_cam_bagros/c1'
2015-10-07 13:55:12: WARNING: control hue not found for device /dev/video0
2015-10-07 13:55:12: ERROR: failed to execute statvfs: [Errno 95] Operation not supported: '/data/media/motioneye_192_168_1_222_cam_bagros/c1'
2015-10-07 13:55:13: WARNING: control hue not found for device /dev/video0
2015-10-07 13:55:13: ERROR: failed to execute statvfs: [Errno 95] Operation not supported: '/data/media/motioneye_192_168_1_222_cam_bagros/c1'
2015-10-07 13:55:15: WARNING: Connect error on fd 14: ECONNREFUSED
2015-10-07 13:55:15: WARNING: Connect error on fd 14: ECONNREFUSED
2015-10-07 13:55:15: ERROR: connection problem detected for mjpg client for camera 1 on port 8765

ccrisan, 2015-10-07:
The only thing I can do for you at this point is to ask you to setup SSH remote access for me and I'd be happy to try to investigate what's going on with your SMB share.

BaGRoS, 2015-10-07:
I can give You access, but how? Only private contact.

ccrisan, 2015-10-07:
Of course, I wouldn't have imagined it otherwise :) Please contact me directly by mail at ccrisan around gmail.

BaGRoS, 2015-10-08:
Specially for testing, I created a new account
login: raspberry password: raspberry
and things started to work.
Following this path changed the account name and password at:
login: RaspberrY and password: RaspberrY
and it does not work connection.