ccrisan/motionpie #330: Request snapshot via motion web interface

Samfox2, 2015-10-07:
Hi, 

I have motioneye running and would like to request a snapshot from motion via its web interface from command line
: 
curl -s -o /dev/null http://piip:8080/0/action/snapshot
but it is not working.

Any idea why, does motioneye block any motion features?
http://www.lavrsen.dk/foswiki/bin/view/Motion/RemoteControlHttpfor details on using http commands to control motion.

ccrisan, 2015-10-07:
Motion control interface port is configured at 7999 and will only listen locally (i.e. on 127.0.0.1), for security reasons. If you want a snapshot, you can get it using the snapshot URL (see the Video Streaming section). For more details, see [this](https://github.com/ccrisan/motionpie/wiki/FAQ#whats-the-deal-with-snapshot-url-streaming-url-and-embed-url) wiki article.

Samfox2, 2015-10-08:
Thanks for your answer. Unfortunately in my case no picture is created. May I ask you to test it with your configuration? Is it possible to disable the "local" listen feature for testing purposes?

ccrisan, 2015-10-08:
I tested it and many users are using it. What URL do you use by the way?

P.S. I'll consider adding an option to enable motion HTTP control listening on all interfaces.

Samfox2, 2015-10-08:
I tried it with:
curl -s -o /dev/null http://localhost:7999/0/action/snapshot but no pic was generated.

ccrisan, 2015-10-09:
You seem to have ignored my answer. I'll try again: please use the Snapshot URL from the Video Streaming section of your motionPie. It should look something like http://ip:port/picture/1/current/?_username=admin&_signature=deadbeef

Samfox2, 2015-10-09:
Hi ccrisan,

thanks for your help (again).
I was not ignoring your answer, I missunderstood it just a little bit.
However, I got it to work now :-)

What I wanted was to take an actual jpg-picture from my doorcam with an user defined event (a signal from my doorbell) while motioneye /-pie is running.

In my case this is:
curl -o "destination-filepath" myip:myport/picture/1/current

Afterwards I send it via telegram to my mobile phone.

Maybe this is also intresting for another motioneye /-pie user...

closed 2015-10-11 by ccrisan