ccrisan/motionpie #331: How to access motionPie with public key authentication?

mastix, 2015-10-07:
Hey ccrisan,

I know that motionPie runs on BuildRoot. So it's a basic linux distro.

I've created my public key on my raspbian jessie system (ssh-keygen) in order to access my motionpie via public key authentication.

Unfortunately I have no idea where to put the authorized_keys file? There's not ".ssh" folder. So I've created one in /home/ftp (which looks like the admin's home folder) and one in /root/ and copied the authorized_keys file there. No luck.

Can you help me out here?

Thank you very much in advance!

Sascha

ccrisan, 2015-10-08:
I never thought one would use SSH to this extent with motionPie :)

If you take a look at `/etc/sshd_config`, you'll see the line `AuthorizedKeysFile /data/etc/ssh_authorized_keys`.

Just place your `authorized_keys` file in `/data/etc`, rename it to `ssh_authorized_keys` and that should do it.

mastix, 2015-10-08:
Oh... sure. Damn it, where was my brain... :D Of course I could have checked the ssd_config! :)

closed 2015-10-08 by mastix