ccrisan/motionpie #334: NFS support

piecam, 2015-10-11:
HI Chris,

Upon my reading, it seems NFS is a bit faster and having lower overhead as remote storage system, comparing to SAMBA. Many offshelf NAS has NFS, as well as SAMBA support.

I was wondering whether you can have it on the GUI as a choice (SAMBA, NFS). 

A typical linux client needs "nfs-common" for nfs4/5 only. A sample command is 
mount /mnt nasip:/folder.

I use autofs like this, (autofs is a bit easier on system with less CPU overhead, comparing to /etc/fstab. It's mounted as needed.)
cat /etc/auto.qnap
media 192.168.100.10:/Recording
cat /etc/auto.master 
/mnt /etc/auto.qnap

Thanks
peng

ccrisan, 2015-10-11:
NFS support has been requested a few times now but is not as important as other features that are already "queued". I'll leave this issue open for now and we'll see if I get some time to implement it in the future.

piecam, 2015-10-11:
HI Chris,

thanks a lot!!
Peng