ccrisan/motionpie #336: Problem with admin credentials

malcolmct, 2015-10-13:
Calin - thanks for doing such a great job with motionpie.  It's an excellent application and for the most part works superbly.

I've had one issue which has affected my installation a number of times and that is the apparent corruption of admin credentials for the web interface.  I can still use my admin password to log into a ssh session as root, but in the course of tweaking settings through the web interface, on each occasion something has happened to affect my admin password and this gives an 'Invalid Credentials' message, preventing me from logging in again via the web.  

This has happened to me now on three separate occasions and I'm afraid I can't pin it down to changing particular settings or to a specific browser (I've used current versions of Firefox(Linux), Chrome(Mac) and Safari(Mac)).   My fix on each occasion has been to re-write the SD card and start again.

Is there a way of resetting the GUI admin password via ssh until this issue is addressed?

Many thanks,

   Malcolm

mikecarr, 2015-10-13:
I have had issues on Chrome (mac) when 1Password was enabled, I now only use Safari without the plugin and I no longer have issues.

malcolmct, 2015-10-13:
Thanks mikecarr - that does seem a likely cause as I have 1Password enabled in my Mac browsers,  I'll test with a browser that doesn't use 1Password and will post any findings.