ccrisan/motionpie #337: Interval Snapshots

drknstrmyknight, 2015-10-14:
When using interval snapshots, neither the retention period nor the working schedule is observed.

ccrisan, 2015-10-15:
The working schedule should only affect the motion detection mechanism, never the media files themselves. This is the way it was designed.

closed 2015-10-15 by ccrisan

drknstrmyknight, 2015-10-15:
Would be nice to have a schedule. It's an outdoor camera and I save 12 hours of blackness...What's the mechanism used for time lapse movie production?

ccrisan, 2015-10-15:
I'll take your request into consideration when rewriting the working schedule part.

The mechanism for time lapse is simply a ffmpeg command that selects pictures from a folder and creates a movie from them with a given frame rate.

drknstrmyknight, 2015-10-15:
I'm saving the time-lapse photos to a folder on the MotionEye Debian machine and I'd like to script creating a video each day and cleaning up the files afterward on that machine rather than burdening the Raspberry with that task.