ccrisan/motionpie #339: Motion movies are laggy

TechGiga, 2015-10-16:


ccrisan, 2015-10-16:
On RPi1 they usually are. Try reducing the frame rate and/or resolution. If that's not an option, buy a more powerful board.

TechGiga, 2015-10-16:
I have an RPi2.

ccrisan, 2015-10-16:
What resolution & what framerate? How laggy are your videos?

TechGiga, 2015-10-16:
640x480 at 24fps. Sometimes the video is not laggy, but every video has no movement between seconds at some point.

TechGiga, 2015-10-17:
Here's an example (at night so as not to reveal my exact location):
https://youtu.be/JHRYIcQzdfY

ccrisan, 2015-10-17:
To be honest, 24 fps is a lot, even for a RPi2. Have you tried reducing the frame rate and see if the problem still persists?

TechGiga, 2015-10-17:
I lowered both the video device and streaming framerate to 15 (although the stream is never more than 1-2fps) and it is better, the moving car actually moves more and jitters less (although it still jitters a bit) but the leaves on the trees are blowing smoothly the whole time (which is good)

TechGiga, 2015-10-17:
When the framerate is at 10, it is not as smooth (of course) but the jitters are more distributed.

ccrisan, 2015-10-19:
You may also try to save the movies on an external HDD or a network drive. Saving them on the SD card can also slow down the entire process if the movies are relatively large.

TechGiga, 2015-10-19:
I regularly download both the movies and pictures to my computer's HDD then delete them from the RPi. The movies are mostly under 3 MB each.

ccrisan, 2015-10-19:
Then I have no explanation.

TechGiga, 2015-10-19:
10fps seems to be the best as there is almost no jittering.