ccrisan/motionpie #341: MotionPie for Raspberry PI 1 - /dev/video0 not available

nosajesahc, 2015-10-19:
I was having a problem getting video0 to appear with the latest version of MotionPie (20151011) for the Raspberry Pi 1 and the Raspberry Pi Camera ( https://www.raspberrypi.org/products/camera-module/ ) . (motioneyeos-raspberrypi-20151011.img.gz)

MotionPie was unable to get video from the camera but raspistill worked fine.

After a lot of digging I found that I needed to have the bcm2835_v4l2 module loaded an in /etc/modules.  I check, it was there as:
```
  bcm2835-v4l2 max_video_width=2592 max_video_height=1944
```
And loaded but not used:
```
[root@meye-65e65822 etc]# lsmod 
Module                  Size  Used by    Not tainted
8192cu                439528  0 
spi_bcm2708             4476  0 
i2c_bcm2708             4344  0 
bcm2835_v4l2           29656  0 
```
Other sites said to add just "bcm2835-v4l2" to modules.  I remounted /dev/root rw then changed the modules file to:
```
  bcm2835-v4l2
```
After a reboot, I was able to add the Raspberry Pi camera to MotionPie and get a clear image.

The module is now in use.
```
[root@meye-65e65822 ~]# lsmod
Module                  Size  Used by    Not tainted
8192cu                439528  0 
spi_bcm2708             4476  0 
i2c_bcm2708             4344  0 
bcm2835_v4l2           29656  1 
```
Hope this helps someone else.

ccrisan, 2015-10-19:
This is not alright. Can you please revert your /etc/modules file back to "bcm2835-v4l2 max_video_width=2592 max_video_height=1944", reboot and see if the problem still persists? The next update will do that anyway.

Normally those are extra parameters that shouldn't affect the presence of /dev/video0. It may have been another issue that got fixed after reboot.

nosajesahc, 2015-10-19:
You're right.  It's working with "bcm2835-v4l2 max_video_width=2592 max_video_height=1944".

Interesting.  I don't think I changed anything else.  I just copied the image onto this SD card today.

```
[root@meye-65e65822 etc]# ls -lat
total 134
-rw-r--r--    1 root     root            57 Oct 19 21:53 modules
drwxr-xr-x   21 root     root          1024 Oct 11 19:19 ../
drwxr-xr-x   10 root     root          1024 Oct 11 19:19 ./
drwxr-xr-x    2 root     root          1024 Oct 11 19:19 init.d/
-rw-r--r--    1 root     root         16253 Oct 11 19:19 ld.so.cache
-rw-r--r--    1 root     root             0 Oct 11 19:19 ld.so.conf
```

Thanks for the quick reply.

closed 2015-10-19 by nosajesahc

ccrisan, 2015-10-20:
It might have been a hardware connectivity problem.

nosajesahc, 2015-10-20:
I think it was.  I found a loose cable on the camera board itself.  It wasn't fully seated.