ccrisan/motionpie #342: Turn OFF webcam's led

surak, 2015-10-21:
Some webcams have a bunch of LEDs that point to the person looking at it. Those less can be enabled/disabled via software

It makes sense for videoconference, but not for surveillance. An option to turn off these leds is desirable.

remonlam, 2015-11-08:
You mean the Pi camera?
That's kinda easy just add this line (disable_camera_led=1) to the /boot/config.txt

iainbryden, 2015-11-12:
Error writing config.txt: Read-only file system

ccrisan, 2015-11-12:
First of all, https://github.com/ccrisan/motionpie/blob/master/README.md.

Secondly, you should enable/disable the camera led from the settings panel, not by manually editing the config files. Mind you that the setting will only work for the Raspberry PI CSI camera.

And lastly, if you really want to manually edit the config.txt file: https://github.com/ccrisan/motionpie/wiki/Tweaks#partition-layout

closed 2015-11-12 by ccrisan