ccrisan/motionpie #346: RPI2 - stopped sending email alerts - odd log entries

rafi1975, 2015-10-27:
Hi,

My RPI2 with the latest motionpie stopped sending email alerts when a motion is triggered. A video is generated when a motion is detected though.
I also noticed some odd errors in motionpie.log as follows:
2015-10-27 00:31:18: WARNING: Write error on <socket._socketobject object at 0x761a2ea0>: [Errno 113] No route to host
2015-10-27 00:31:18: ERROR: exception calling callback <bound method HTTP1Connection._on_write_complete of <tornado.http1connection.HTTP1Connection object at 0x761007f0>> for <tornado.concurrent.Future object at 0x761004f0>
Traceback (most recent call last):
  File "home/ccrisan/motionpie/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/concurrent.py", line 290, in _set_done
  File "home/ccrisan/motionpie/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/http1connection.py", line 462, in _on_write_complete
  File "home/ccrisan/motionpie/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/concurrent.py", line 209, in result
  File "<string>", line 3, in raise_exc_info
error: [Errno 113] No route to host
2015-10-27 02:06:36: ERROR: HTTP 404: Not Found (not found)
2015-10-27 02:06:36: WARNING: 404 GET /cgi-bin/bash (70.129.223.20) 5.89ms
2015-10-27 02:07:07: ERROR: HTTP 404: Not Found (not found)
2015-10-27 02:07:07: WARNING: 404 GET /cgi-bin/php (70.129.223.20) 5.30ms
2015-10-27 02:07:08: ERROR: HTTP 404: Not Found (not found)
2015-10-27 02:07:08: WARNING: 404 GET /cgi-bin/php4 (70.129.223.20) 3.50ms
2015-10-27 02:07:12: ERROR: HTTP 404: Not Found (not found)
2015-10-27 02:07:12: WARNING: 404 GET /cgi-bin/php5-cli (70.129.223.20) 5.79ms
2015-10-27 02:07:14: ERROR: HTTP 404: Not Found (not found)
2015-10-27 02:07:14: WARNING: 404 GET /cgi-bin/recent.cgi (70.129.223.20) 4.06ms
2015-10-27 02:07:14: ERROR: HTTP 404: Not Found (not found)
2015-10-27 02:07:14: WARNING: 404 GET /cgi-bin/sat-ir-web.pl (70.129.223.20) 3.58ms
2015-10-27 02:07:15: ERROR: HTTP 404: Not Found (not found)
2015-10-27 02:07:15: WARNING: 404 GET /cgi-bin-sdb/printenv (70.129.223.20) 3.42ms
2015-10-27 02:07:22: ERROR: HTTP 404: Not Found (not found)
2015-10-27 02:07:22: WARNING: 404 GET /cgi-bin/test-cgi.pl (70.129.223.20) 5.84ms
2015-10-27 02:07:23: ERROR: HTTP 404: Not Found (not found)
2015-10-27 02:07:23: WARNING: 404 GET /cgi-bin/test.sh (70.129.223.20) 5.67ms
2015-10-27 02:07:24: ERROR: HTTP 404: Not Found (not found)
2015-10-27 02:07:24: WARNING: 404 GET /cgi-bin/tools/tools.pl (70.129.223.20) 3.56ms
2015-10-27 02:07:28: ERROR: HTTP 404: Not Found (not found)
2015-10-27 02:07:28: WARNING: 404 GET /cgi-sys/defaultwebpage.cgi (70.129.223.20) 6.85ms
2015-10-27 02:07:32: ERROR: HTTP 404: Not Found (not found)
2015-10-27 02:07:32: WARNING: 404 GET /cgi-sys/entropysearch.cgi (70.129.223.20) 4.62ms
2015-10-27 02:07:33: ERROR: HTTP 404: Not Found (not found)
2015-10-27 02:07:33: WARNING: 404 GET /cgi-sys/php5 (70.129.223.20) 5.55ms
2015-10-27 02:07:33: ERROR: HTTP 404: Not Found (not found)
2015-10-27 02:07:33: WARNING: 404 GET /phppath/cgi_wrapper (70.129.223.20) 9.10ms
2015-10-27 02:07:34: ERROR: HTTP 404: Not Found (not found)
2015-10-27 02:07:34: WARNING: 404 GET /phppath/php (70.129.223.20) 3.40ms

ccrisan, 2015-10-27:
1) Your pi appears to be directly connected to the internet or at least having port 80 forwarded to it. It's advisable to use a different port to prevent unwanted http requests like you seem to receive. They'll only eat up resources.

2) I saw a host unreachable error which indicates an internet connectivity error. That might be the cause of your missing emails. The motion.log also contains useful details regarding email sending.

3) You should switch to [motionEyeOS](https://github.com/ccrisan/motioneyeos), as motionPie is discontinued.

rafi1975, 2015-10-27:
Hi Calin,

Many thanks for your prompt and detailed reply.

I'll check what you suggested.
Is it possible to upgrade from motionPie to motionEyeOS? The reasom I'm
asking is that I have the camera installed outside of the house on a wall
and I have to disassemble it completely to reach SD card.

Thanks again,

Rafal

ccrisan, 2015-10-28:
https://github.com/ccrisan/motionpie/blob/master/README.md

closed 2015-10-28 by ccrisan

rafi1975, 2015-10-28:
Ok. Fair enough. Install from scratch then.

Thanks