ccrisan/motionpie #347: building from source for more functionality for rpi2

ivr588, 2015-10-30:
Hi,
Your software works just great out-of-the-box with rpi2 and piNoIR camera. My only issue is saving the output offsite. I followed the building-from-source section of your wiki and in the xconfig section I selected sshfs (and fusefs because sshfs depends on it). I was careful not to change anything else. It compiled nicely and built the correct image file for raspberry2. Transferred it to the microsd card but unfortunately it wouldn't boot. I would be very grateful if you would answer the following, hopefully I'm not doing something silly:

1. Is this the best way to go about enabling sshfs? 

2. As well as the image motioneyeos-raspberrypi2.img, I also have the following files in the images directory:
boot.img
rootfs.tar
root.img
zImage
Are any of these needed? The reason I'm asking is because my image didn't even boot.

3. will writeimage.sh work as intended with the built-from-source image?

Many thanks

ivr588, 2015-10-30:
Solved this now. The reason the image didn't work before was because there was a problem with the microsd card. What I did to fix it:
1. full format that took ages with a sd card formatting utility
2. set IP settings etc with writeimage & wrote the image
3. booted up & ssh into the pi
4. mount -w / -o remount
5. ssh-keygen -t rsa -b 4096 [accept the defaults and use an empty passphrase]
6. mount -r / -o remount
6a. mkdir /data/remote
7. cd /root/.ssh
8. sftp me@myremote.host
9. put id_rsa.pub authorized_keys [and exit from sftp]
10. sshfs me@myremote.host:/full/path/to/storage /data/remote -o reconnect

now need to make modprobe fuse persist and a string for fstab so sshfs auto-starts on boot

ccrisan, 2015-10-30:
Yeap, these are the correct steps. I'm glad you managed to make it work.

closed 2015-10-30 by ccrisan