ccrisan/motionpie #348: Ratio Issue at 2592 x 1944

GizmoRA2, 2015-10-30:
Hi, I am having an issue getting the correct ratio with the noir camera setup to run at 2592 x 1944

I copied the code from https://github.com/DavidLutton/MotionEye/blob/master/src/v4l2ctl.py
I then pasted it into a new file at /usr/lib/python2.7/site-packages/motioneye/v4l2ctl.py
It added the new resolution, however as I understand it 2592 x 1944 is supposed to use the entire sensor, 1280 x 1024 seems to, and it shows a nice wide and tall image, however 2592 x 1944 seems to crop top and bottom as well as left and right and I am unsure why that would be happening

Many thanks
Gizmo

GizmoRA2, 2015-10-30:
I have just realised this maybe posted in the wrong place as it relates to motionpieos
My appologese if this is the case, as I am not sure if it is specific to motionpieos or if it is more relevant to motionpie as the underlying software

Also this is running on a RPi B if that makes any difference though I am sure it wont :)

GizmoRA2, 2015-10-31:
1280x960 seems to yield the best results so far and works with fast network camera mode

However it would be nice to get a higher resolution at the 4:3 ratio working (taking advantage of the entire sensor)

ccrisan, 2015-10-31:
Both the width and the height must be *modulo 16* numbers in order to work with *motion*. Cropping may occur due to using the video port instead of the still images port of the CSI camera. However I strongly suggest not using resolutions that high with motionEye(OS) as they generate way too much data for the PI to process, even at lower framerates.

More details on this matter should be found on the Raspberry PI forum.

GizmoRA2, 2015-10-31:
Ok

I might just use stills for push notifications :)

Thanks for the quick reply
Gizmo

closed 2015-10-31 by GizmoRA2