ccrisan/motionpie #352: Email being sent without attachments

mikecarr, 2015-11-16:
After a while of working perfectly I started to receive emails with no attachments. The only error I see in the logs is the following:
````
2015-11-16 08:51:24: [motioneye]    ERROR: failed to create movie preview for /data/output/2015-11-15/21-48-01.avi: Command 'ffmpeg -i "/data/output/2015-11-15/21-48-01.avi" -f mjpeg -vframes 1 -ss 4 -y /data/output/2015-11-15/21-48-01.avi.thumb' returned non-zero exit status 1
````

I still have plenty of space
````
Filesystem                Size      Used Available Use% Mounted on
/dev/root               174.3M     92.3M     69.4M  57% /
devtmpfs                429.1M         0    429.1M   0% /dev
/dev/mmcblk0p1           19.9M      7.3M     12.7M  36% /boot
/dev/mmcblk0p3           28.9G    745.1M     26.7G   3% /data
tmpfs                   433.2M         0    433.2M   0% /dev/shm
tmpfs                   433.2M      3.0M    430.2M   1% /tmp
/dev/mmcblk0p3           28.9G    745.1M     26.7G   3% /home/ftp/sdcard
/dev/mmcblk0p3           28.9G    745.1M     26.7G   3% /home/ftp/storage
````

If I run the command on the command line I get the following error
```
[avi @ 0xb5e350] Format avi detected only with low score of 1, misdetection possible!
/data/output/2015-11-15/21-48-01.avi: Invalid data found when processing input```

ccrisan, 2015-11-16:
The movie files have nothing to do with email attachments. If you open the pictures media browser, do you see pictures taken around the time of the motion events that are notified by email?

mikecarr, 2015-11-16:
Yes, avi and jpg

```
-rw-rw-rw-    1 root     root      639.7K Nov 16 11:21 11-20-58.avi
-rw-rw-rw-    1 root     root       74.0K Nov 16 11:20 11-20-58.jpg
-rw-rw-rw-    1 root     root       74.2K Nov 16 11:21 11-20-59.jpg
-rw-rw-rw-    1 root     root       74.4K Nov 16 11:21 11-21-00.jpg
-rw-rw-rw-    1 root     root       74.1K Nov 16 11:21 11-21-01.jpg
-rw-rw-rw-    1 root     root       73.9K Nov 16 11:21 11-21-02.jpg
-rw-rw-rw-    1 root     root       74.5K Nov 16 11:21 11-21-03.jpg
-rw-rw-rw-    1 root     root       74.6K Nov 16 11:21 11-21-04.jpg
-rw-rw-rw-    1 root     root       73.6K Nov 16 11:21 11-21-05.jpg
-rw-rw-rw-    1 root     root       72.8K Nov 16 11:21 11-21-06.jpg
-rw-rw-rw-    1 root     root       53.5K Nov 16 11:21 11-21-07.jpg
-rw-rw-rw-    1 root     root       53.9K Nov 16 11:21 11-21-08.jpg```

ccrisan, 2015-11-17:
I got and read your logs. The problem is that you have too many media files and, by the time motionEye needs to send the email notification it simply takes too much to list them:

    2015-11-13 21:14:52: [sendmail]    ERROR: timeout waiting for the media listing process to finish

In this case, the sendmail mechanism decides to send the mail right away (without attachments), given the importance of the notification. Please remove older media files and see if the problem persists.

mikecarr, 2015-11-17:
deleting files fix it for now, I switched the photos to store for one week
as opposed to one month. Will see if that solves it going forward.

Thanks

b0rnwr0ng, 2016-05-20:
Sorry for coming back to such an old thread, but is it possible to create an option for the Email to wait xx seconds before sending an Email?

ccrisan, 2016-05-20:
@b0rnwr0ng I hope you have already switched to [motionEyeOS](https://github.com/ccrisan/motioneyeos).

You could either play with [list_media_timeout_email](https://github.com/ccrisan/motioneye/wiki/Configuration-File#list_media_timeout_email) if you're concerned with the amount of time it takes to gather the attachments. Or you could play with the `Attached Pictures Time Span` option in the settings menu if you just want to delay the email sending in order to gather more picture.

b0rnwr0ng, 2016-05-20:
I did install the MotionEyeOS, I realized after I posted it was in regards to the older app.

Giving the list_media_timeout_email a try in the motioneye.conf file -- I'll keep you posted.  Thanks for your help!