ccrisan/motionpie #353: Initial Boot asks for login

akapacker, 2015-11-18:
Hi Calin,
I'm having trouble getting MotionPie working on my new Raspberry Pi 2 Model B v1.1.
I downloaded the image for RPi2: motioneyeos-raspberrypi2-20151103.img.gz
from https://github.com/ccrisan/motioneyeos/releases
Unzipped it with 7zip.
Formatted a microSD card.
Used Win32DiskImager on my Windows 7 laptop to write the image to the microSD card.
Inserted the microSD card and boot up the RPi.
And then the initial Boot stops at:
"Welcome to motionEye!
meye-0c56ed06 login:"

Not sure what to login with.
I suspect that I'm not doing something right, because I've googled this problem heaps of times/combinations, and cannot find someone who has had a similar problem.
Could you help me out please? 
Thanks for your time!
Cheers
David

ccrisan, 2015-11-18:
https://github.com/ccrisan/motioneyeos/wiki/FAQ#i-am-only-getting-white-on-black-text-on-my-monitor-where-is-my-ui

closed 2015-11-18 by ccrisan

stavropn, 2016-04-25:
Sorry but it is valid!

ccrisan, 2016-04-25:
https://github.com/ccrisan/motioneyeos/wiki/FAQ#command-line--shell-access

akapacker, 2016-04-29:
Yeah, I couldn't get it to work, so I gave up. Thanks for trying though!

ccrisan, 2016-04-29:
@akapacker what were you trying to do, more precisely? Did you try `admin` with no password? Why were you trying to log in to the console anyway?

akapacker, 2016-05-03:
Hi Calin,
Sorry for the late reply. It was so long ago, I've forgotten. But, I will
have another go at it, and let you know. Thanks for replying. Very
responsive. I am most grateful.
Cheers
David