ccrisan/motionpie #354: Windows share not working

killjaysinger, 2015-11-18:
This is an open share with "everyone" on my windows network. I can access it fine from another windows machine:

[image]

However motionPie will not save images or videos to this location:

[image]

what am I missing?

ccrisan, 2015-11-19:
It's hard to say what the problem is. Is the share public (I assume that's what "shared with everyone" means). If yes, why have you set a username and password?

I really need to add a test button when configuring the SMB shares.

drknstrmyknight, 2015-11-19:
Are you using a fully qualified, resolvable name or IP address for the server ?

killjaysinger, 2015-11-19:
Yes it is a public share. I put the user and password in there because I thought the problem might be an authentication issue so I added a user that I know has full access (the local admin) to the account. I've tried it with both blank also with no results. I've also tried just using the ip address rather than just the name. 

I've also noticed at times when there is motion in front of the camera that the border around it turns red. Is that typical for the motion detection?

ccrisan, 2015-11-19:
The reddish frame is indeed an indication of motion detection.

killjaysinger, 2015-11-19:
Are there any log files I can look at to see what the problem might be with it saving to a windows share? I feel certain it's setup correctly because I can access it from other machines

ccrisan, 2015-11-20:
There's nothing interesting in that log, apart from a "permission denied" error message. Do you by any chance have another Linux machine around?

killjaysinger, 2015-11-20:
I can probably setup an old mac

geeksteev, 2015-12-22:
I'm having the same issue. Looking at motioneye.log shows that there's a failure mounting the share. Is this supposed to be using a cifs share or smb? It would also be extremely helpful to incorporate a new line function in the logs for readability. This is the output of motioneye.log:

    2015-12-22 17:17:06: INFO: interrupt signal received, shutting down...
    2015-12-22 17:17:06: INFO: server stopped
    2015-12-22 17:17:07: INFO: interrupt signal received, shutting down...
    2015-12-22 17:17:30: INFO: hello! this is motionPie 20150719
    mount error(13): Permission denied Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
    mount error(22): Invalid argument Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
    mount error(13): Permission denied Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
    2015-12-22 17:17:31: ERROR: failed to mount smb share "//192.168.1.118/security" at
    "/data/media/motioneye_192_168_1_118_security_security"
    2015-12-22 17:17:31: INFO: cleanup started
    2015-12-22 17:17:31: INFO: wsswitch started
    2015-12-22 17:17:31: INFO: thumbnailer started  
    2015-12-22 17:17:32: INFO: server started
    2015-12-22 17:18:34: WARNING: control hue not found for device /dev/video0
    2015-12-22 17:18:34: WARNING: control hue not found for device /dev/video0
    2015-12-22 17:18:34: WARNING: control hue not found for device /dev/video0
    2015-12-22 17:18:34: WARNING: Connect error on fd 14: ECONNREFUSED
    2015-12-22 17:18:35: WARNING: Connect error on fd 14: ECONNREFUSED
    2015-12-22 17:18:35: ERROR: connection problem detected for mjpg client for camera 1 on port 
    8080

ccrisan, 2015-12-23:
Are you sure you have correctly identified and configured the network share fields (server, share, username, password, root directory).

geeksteev, 2015-12-23:
I'll do some trouble shooting and validate.  I'll keep you posted.

geeksteev, 2015-12-23:
After some trouble shooting, I'm still receiving the same errors. I've been
able to connect to the file share(Windows) via Windows, OSX and Linux
clients. I'm currently using the file server IP address as to rule out a
possible DNS issue. I've attached a copy of the console output so you can
see the errors post motioneye starting.

geeksteev, 2015-12-23:
I'm also curious as to why I'm getting cifs errors if motioneye is
connecting using smb.

ccrisan, 2015-12-24:
Would you be able to log remotely via SSH and run some commands? If yes, could you run

    mkdir -p /data/media/test
    mount.cifs //tempimage/security /data/media/test -o username=administrator,password=[password],sec=[none,ntlm or ntlmv2],

trying each of the listed security modes?

killjaysinger, 2016-02-04:
**UPDATE**

In reference to the original problem I was having: I created another shared folder on my network, this time instead of a windows machine I used a system running Linux Mint and it worked fine. Still not sure why it wouldn't work on the windows machine but this seems to be a suitable alternative.

geeksteev, 2016-02-04:
My apologies, I thought I had replied back in December. The issue I found I was having had to do with the password complexity. I was using a complex password to connect to the Windows share. When I used something simple like 'password' with no special characters, it started working.

killjaysinger, 2016-05-23:
I just setup another system with a model b pi. I noticed on this one that the date and time stamp was wrong. I haven't had this issue before. Is there any way to go in and set the date and time manually? 

thanks

ccrisan, 2016-05-24:
@killjaysinger what version? You can't set the date/time manually. It's always updated from the network (normally via HTTP).

killjaysinger, 2016-05-24:
It's the latest version

ccrisan, 2016-05-24:
@killjaysinger and what is that version? Given that you have opened the issue on motionPie (which has been [discontinued](https://github.com/ccrisan/motionpie/wiki) for months now) instead of [motionEyeOS](https://github.com/ccrisan/motioneyeos) makes me think that you're still on motionPie.

killjaysinger, 2016-05-24:
I may not have posted my question in the appropriate forum thread. The release version is 20160410

ccrisan, 2016-05-25:
@killjaysinger if your date is wrong, make sure your system has internet access (DNS and gateway settings are correct). If they are, please attach your boot.log file here.