ccrisan/motionpie #355: MotionPie only streaming video

mvittorini, 2015-11-20:
Hello everyone.
First excuse me for my bad english.
I'm very inexperienced about linux and I bought my first raspberry pi model 2 b board, a few days ago, on which I installed the csi camera and motionpie.
My aim is to achieve a ipcam that produces only a H.264 video stream with audio, to be recorded on my Synology NAS.
The NAS is already connected with an Axis IP camera that produces an H.264 video stream. The NAS is configured to pick up the video stream using the motion detection function of ip camera, rather than the motion detection engine that is on the NAS. No viedo i recorded on the ip camera sinche the AXIS does not has any local storage option.
This works pretty well, except for some temporary loss of connection (less than 5 disconnect - connection at day, but each event lasts only a few seconds) between the NAS and the IP camera, which seems to be due to a bug in the operating system of the NAS.
I want to run my raspberry/motionpie system alike.
That is, I do not need that raspberry registers video on the sd card, only that it produces the H.264 video stream.
In particular I configured motionpie to have a video stream at 1280x720 pixels with 25 fps frame rate; obviously the on-board motion detection is active with a pre-recording= 200 frames and a post recording=50 frames.
Looking in the web interface of motionpie I could not find a way to inhibit the video recording, leaving active the video streaming.
Searching on the web I could not find any help to my problem.
How is it possible to achieve this result?
Thank you so much.

mvittorini

ccrisan, 2015-11-20:
First of all, [motionPie is obsolete](https://github.com/ccrisan/motionpie/blob/master/README.md). You should switch to [motionEyeOS](https://github.com/ccrisan/motioneyeos).

Second of all, motionPie/motionEyeOS/motion cannot stream H.264, it only speaks MJPEG.

And lastly, if you'll start to use motionEyeOS, you'll see that there is a "Movies" option that can be disabled (and is disabled by default).

mvittorini, 2015-11-20:
Thank you very much for your fast reply Calin.
There is any guide for dummy that explayns how to create an image of motioneye os for the raspsbery pi 2?

mvittorini, 2015-11-20:
Do not worry i found the wiki:
https://github.com/ccrisan/motioneyeos/wiki
Thanks a lot.

mv