ccrisan/motionpie #358: attached camera can't be detected

selfcommander, 2015-11-25:
I see light on the camera. But MotionPie can't detect it. Attached is the screenshot of what I am seeing in web-interface, how it looks from outside and logs

http://s000.tinyupload.com/?file_id=54866614851102141702

ccrisan, 2015-11-25:
Is this a clean install or did it happen after playing with some settings?

selfcommander, 2015-11-25:
Clean

selfcommander, 2015-11-25:
Also tried previous builds, with clean installs. Still can't find it

ccrisan, 2015-11-25:
1) Your dmesg.log file says `[    4.524768] : bcm2835-v4l2: error -1 while loading driver`, which indicates a problem with your camera device. Check the ribbon cable connection, make sure you have at least 96 MB of RAM allocated to the GPU and if it still fails, try Raspbian and see if it works that way. If it doesn't, replace your camera board.

2) You should consider switching to [motionEyeOS](https://github.com/ccrisan/motioneyeos), as motionPie is discontinued and will not receive improvements and fixes anymore.

selfcommander, 2015-12-05:
Camera was plugged into the wrong port -_-
The guy who owned it previously left it plugged in there, so I assumed it
is the right one...
Thanks for letting me know about new OS

closed 2015-12-06 by ccrisan