ccrisan/motionpie #362: Sharing from MotionPie to Windows PC

Castor66, 2015-12-12:
Hello everyone,

I think I'm less technically proficient than most of you, so I will ask for help on my problem.  I am trying to set up my MotionPie to store files through my network to a Windows PC.  I don't know enough about sharing to do it properly.

On my PC, which is titled "LOFT-MINI", I have created a folder called "RaspberryPiSecurity" and enabled sharing with read/write access to Everyone on it.

Here is how I have set up the Storage Device section of settings, yet no files seem to be saving to this folder:
Storage Device: Network Share
Network Server: 192.168.1.2 (my computer's IP address)
ShareName:  FrontDoorCam (just a name I gave it, not sure if that's what I'm supposed to do)
Share Username:  blank (I didn't put anything here because I don't think I need to since I told my Windows share to allow Everyone to read/write to the folder)
Share Password: blank also
Root Directory:  /LOFT-MINI\RaspberryPiSecurity (again, I'm not at all sure if I'm doing this part right)

Am I missing something?  What here do I need to change, and/or do I need to do on the Windows side to allow this to work?

Thanks,
Josh

ccrisan, 2015-12-13:
`RaspberryPiSecurity` is the Share Name and Root Directory can be left empty (defaulting to /).

Castor66, 2015-12-13:
That worked!  This program is awesome.  I cannot compliment you enough on it.  I have two more questions for you if that's okay.  First, how do I make the video feed/pictures accessible from a phone/computer that is off the network?  I see the address the program provides in the Streaming URL field, but will that work when I'm not locally on the network?  Secondly, does the email notification functionality have a way to send text messages too?  Or just emails?  Thanks so much!

ccrisan, 2015-12-13:
1) Accessing a device on your local network from the Internet is something that has to do with your network provider and your router. It has nothing to do with motionEye/Pie/OS. In theory you should have a public IP address from your ISP and forward ports 80 (and optionally 8081) to your PI.
2) No. SMS (if that's what you were hoping for) is not supported.

closed 2015-12-13 by ccrisan

Castor66, 2015-12-13:
Thanks, that helped.  I found Port Forwarding instructions and got it to work.  As someone not quite as technical as everyone else on here, this is very exciting to successfully get working.  Thanks for your personal help!