ccrisan/motionpie #363: Camera not detect

haviana, 2015-12-19:
Hi there,

I heard about your project and i felt curiosity for testing it. So, i brought an raspberry pi camera surveillance and then install motion on it. I have the motionpie correctly installed and everything seems to work fine. However, yesterday i check the camera and RPI2 can not recognise the camera. I already do the following:
 1. Unplug and plug again the cable which connects the raspberry with the camera.
 2. Reboot
 3. Unplug and plug the power cable.

In attachment you can find the logs files.

Thanks in advance.
[log.zip](https://github.com/ccrisan/motionpie/files/67492/log.zip)

ccrisan, 2015-12-20:
Hi,

a) So your camera worked with the RPI2 and motionPie before but suddenly stopped working? If yes, please try to see if it works with Raspbian, so we know whether it's a motionPie-related issue or not.

b) I'd need to see your dmesg.log file (downloadable from the UI)

c) You should consider switching to [motionEyeOS](https://github.com/ccrisan/motioneyeos) as motionPie is no longer developed.

ibrahimma2016, 2016-02-23:
Hi there
I havealmost the same problem with my new RPi camera module. motionPi cannot detect the camera module automatically as what it did with webcam. The webcam worked nicely with me.

Any suggestions to how enable the camera module in motionpi+Raspberrypi(1) please 

Thanks in advance,
Ibrahim

ccrisan, 2016-02-23:
The RPi camera module should be detected automatically at a fist clean boot. If it doesn't, check the ribbon cable. To make sure the camera module actually works, try it with Raspbian.

ibrahimma2016, 2016-02-24:
You're absolutely right.
Thank you so much 

I'v moved to RPi2 board with the same Ethernet cable I had used with RPi1. I thought IP address will be the same with the new board (in browser) but I was wrong hhhh which caused some delay.

Anyway, now it is working:)

ibrahimma2016, 2016-02-24:
I tried to save detected motion (picture) in windows 7 but with no success, can you help me with this issue please

My faild try was:
1) creating a folder with X-name in windows 
2) pasting the folder directory at the "Root Directory" space in motioneyeOs 
3) did apply and reboot 

Are these the true method to save the detected motions?

Thank you

ccrisan, 2016-02-24:
I believe you're trying to use a network share. See this: https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#network-shares.

closed 2016-02-24 by ccrisan