ccrisan/motionpie #364: I can no longer connect to camera

TechGiga, 2015-12-22:
So for some reason, the raspberry pi shows up on the router page as online, but I cannot connect to it via web browser or SFTP. I don't know what to do.

ccrisan, 2015-12-22:
Does it respond to ping? You'll need to make sure the network connection is still on for your PI. If you have a Linux PC, you could insert the SD card into your PC and recover the log files to see what was going on. Also, you might want to switch to [motionEyeOS](https://github.com/ccrisan/motioneyeos).

TechGiga, 2015-12-22:
Pinging it does not work. I think I will put motionEyeOS on it and start fresh.

nickjdunn, 2016-03-29:
I had alot of issues using the pi camera module as a Fast Network Camera and using another pi as the server.  The server would constantly loose connection to the camera, and when I would ping the pi with the camera it wouldn't respond and when I accessed it from a browser I got a red outlined rectangle with no GUI.  Tried restarting and later a fresh format on both the server and the pi with the camera with every attempt resulting in a working setup for a few hours then it would loose connection and not want to work.  Im not sure if this was an issue of having the pi hooked up wireless because I never tried it wired but now I am running a pi3 as the server and bought a couple wired ip cameras and the setup has been working for 3 days not one problem.

I am using the Crenova QD520 and for $25 it is worth it!

http://www.amazon.com/Crenova%C2%AE-Waterproof-iCloud-Security-Outdoor/dp/B00LXS12O6/ref=sr_1_1?ie=UTF8&qid=1459265071&sr=8-1&keywords=crenova+ip+camera

Is there a section of the site where we can see and submit setup instructions for particular cameras?  I know it took me a good hour to figure out the proper setup to get this camera working with MotionEye.

ccrisan, 2016-03-29:
There's no such site but you could always create an [issue](https://github.com/ccrisan/motioneyeos/issues) that is specific to your camera and people can discuss all related matters there.