ccrisan/motionpie #366: Using both LAN and Wireless

therealwave, 2015-12-23:
I cant seem to get both ports working.
My setup:
 I have a Pi2 with a wireless card. I have that working fine. I can connect and configure motionpie on wifi. I have a network (ethernet) ip camera I want to plug into the Pi2 using ethernet port.

If I am doing everything on ethernet using a switch and no wireless it all works fine with the camera.

I have my camera set to 192.168.1.200
Motionpie is 192.168.1.121

Is there a way to get Pi2 wireless connected (192.168.1.121) and have the Motionpie connect to my camera ethernet (192.168.1.200) at the same time?
Thanks,
Brian

therealwave, 2015-12-23:
Doh, Im using motioneyeos. Ill post in that github

ccrisan, 2015-12-23:
motionPie/motionEyeOS was not designed to work with two internet connections at the same time. The only way to obtain what you need is to connect both your PI and your netcam(s) to your local router.

closed 2015-12-23 by ccrisan