ccrisan/motionpie #369: Third camera no longer working

mdobres, 2016-01-20:
Hi, I have a Raspberry PI2, connected to three web cams using the latest version of motioneyeos. When first set up it showed three cameras. Now only two. I've tried re-booting, If I try reconnecting third camera while live, doesn't find camera. Tried increasing GPU to 512 and using PI2 Overclocking.
Any other suggestions. See http://projects.fortherapy.co.uk/sample-page/raspberry-pi-birdbox/

Everything working great with 2 cameras.
Regards

Max

ccrisan, 2016-01-20:
I hope you're really using [motionEyeOS](https://github.com/ccrisan/motioneyeos) in spite of opening the issue for motionPie.

I didn't really get whether rebooting solves the problem or not. I only understood that reconnecting the camera while live (w/o rebooting?) doesn't work. What do you mean by *not showing up*? Does it turn grey? Does the corresponding frame simply disappear from the UI?

You never mentioned anything about the cameras you're using. Are they all USB cameras? What maker/model? Users have been reporting issues when using multiple same model Logitech cameras, for example.

Also please attach all of your log files so we can see what's actually going on.

mdobres, 2016-01-20:
Re-Booting does not solve the problem. Whatever I do I only see two cameras. Yes I am using motionEyeOS sorry for posting in wrong place. Had same problem with motionpie so upgraded.
Log Files attached. Had to rename as .txt 
[boot.txt](https://github.com/ccrisan/motionpie/files/98171/boot.txt)
[motioneye.txt](https://github.com/ccrisan/motionpie/files/98172/motioneye.txt)
[motion.txt](https://github.com/ccrisan/motionpie/files/98174/motion.txt)
[messages.txt](https://github.com/ccrisan/motionpie/files/98173/messages.txt)
[dmesg.txt](https://github.com/ccrisan/motionpie/files/98175/dmesg.txt)

All three cameras are identical USB cameras bought on EBAY. Both the working ones show up as VML2 Camera.
Camera 1 has device: 	
/dev/v4l/by-id/usb-AVEO_Technology_Corp._USB2.0_Camera-video-index0
Camera 2 has device : 	 /dev/video0

The third camera first showed up greyed out, but now it doesn't show at all.no frame nothing and when I try to "Add Camera" there is no camera found.

Thanks for your help.

mdobres, 2016-01-20:
Sorry that should be V4L2 camera not VML2

ccrisan, 2016-01-21:
The power supply unit might not be enough for your scenario. Please try using a powered USB hub.