ccrisan/motionpie #370: motion detection parameters

landru500000, 2016-01-22:
When I first installed motioneyeos  (camera looking through the window to the street), the system detected cars passing by(in front of the condo).  Now, it does not any more, unless the car is being driven at very slow speed.  When I change the sensitivity parameter to less than 3%, I get a lot of video recordings which are false detections.  What am i doing wrong?  the motion detection is set for 24 hours a day every day.

ccrisan, 2016-01-23:
What's the value of your Light Switch Detection option?

landru500000, 2016-01-25:
Hi Calin,

light switch detection set to 0.

key parameters"
video resolution:  320 X 240
still images:  off
video stream: on
movies: on
record mode: motion triggered
frame change treshhold:  5%
auto noise detection: off
noise level: 2%
light switch detection:  0
max movie length 222
motion gap:  4
captured before 3
captured after 4
min motion frames 33

motion notification off

hardware:
raspberri pi 2B
32 gig sds
os:  motioneyeos
number of camers:   2 (usb)

view area: front yard with shrub that are not moving at all.

I tried to see if the frame change treshold is too low.  I set it at high
levels,, and got the same results.  very frequent video recording, with
nothing really moving.

thank you very much  in advance for your help.

ccrisan, 2016-01-25:
Enabling auto-noise detection might help.