ccrisan/motionpie #376: MJPEG stream with UI?

ninninon, 2016-02-07:
Hi, is there any way of streaming the footage from the pi using port forwarding (on a different network) but still have all the controls? If this isn't already implemented then it'd be very useful :)

ccrisan, 2016-02-08:
Yes. Enable port forwarding from port e.g. 8888 to port 80 on your router. You'll then be able to access the UI from everywhere. Additionally, if you have enabled Fast Network Camera, you'll need to forward port 8081 to 8081 as well.

closed 2016-02-08 by ccrisan

ninninon, 2016-02-08:
I'm currently it as a simple MJPEG streamer with basic security so I have a password on it. So do you mean that in my router I would port forward 8081-8081 instead of just 8081, and then type xx.xxx.xx.xx:8081-8081 instead of just xx.xxx.xx.xx:8081? I'm pretty new to port forwarding :P

ccrisan, 2016-02-09:
I'm not sure I get your problem. If you have Fast Network Camera enabled, you'll need to forward incoming port 8081 to port 8081 on your internal IP address. The syntax "8081-8081" is not valid for a port specification. If you don't have Fast Network Camera enabled, you don't need to care about port 8081 at all.

ninninon, 2016-02-13:
I'm using fast network camera, and when I go to networking section, it shows my port is 8081, there is no mention of any other port, it's just when I use that I can see anything other than a stream, I can't change the settings- there's no UI. However, under expert settings it does say that the https port is 80...