ccrisan/motionpie #377: Motion sensor recording OFTEN

androidshrink, 2016-02-12:
Got setup with motion sensor and it appears with some cameras there is a frequent video feed grafting or rearranging of the image. This is setting off the motion capture and resulting in a short video almost every minute or multiple ones in every minute. please advise. I have tons of video footage ;) [<iframe width="560" height="315" src="https://www.youtube.com/embed/gEMEiTCuzgw" frameborder="0" allowfullscreen></iframe>](url)

androidshrink, 2016-02-22:
I had a small cheap Microsoft webcam setup and it didn't have any of the motion capture fragmentation and false recordings as mentioned above, but I have tried 2 additional higher end usb cameras and getting the same graphical fragmentation that results in a false motion capture. has no one ever had this happen before? tried changing things around like frame rate or aspect ration and seems like it occurs less often at lower frame rates and lower aspect ratios but not sure.

ccrisan, 2016-02-23:
Unfortunately this is something that only motion developers can fix or do something about. 

P.S. please switch to [motionEyeOS](https://github.com/ccrisan/motioneyeos)

closed 2016-02-23 by ccrisan